mabuchilab/Instrumental #11: Add Unit Testing

natezb, 2016-03-05:
This one's challenging because Instrumental is so hardware-centric, and failure to import unsupported drivers is actually a feature. Also, it's hard to automatically (from software) tell that hardware is working correctly, and you don't want to tell an instrument to start doing random things like setting DAQ voltages that could break something.

However, you could have optional hardware tests that do simple things like opening and closing a device, maybe querying some basic info. Non-hardware, non-optional codepaths should be as fully tested as possible.

At there very least, there should be regression testing to make sure drivers don't break due to stupid things like syntax errors that were overlooked.

After looking into the frameworks, I'd like to use `pytest`.

goerz, 2016-03-07:
I'd definitely recommend `pytest`. For testing against hardware interfaces, I'd suggest trying to encapsulate the communication as much as possible in some sort of wrapper routine. For testing, you could then either use the Mock library to replace that routine with a dummy, or you could use a "record & replay" model, where you run the test against the actual hardware and record all the communication to a file, and then have a flag in your test so that you can get all the responses from the recorded file instead, and there's no actual communication with the hardware.

I've had a similar situation recently where I needed to test the communication with remote HPC cluster schedulers, where I couldn't assume that I would be able to actually connect the the clusters during the test. In that case I solved the problem by encapsulating system calls in a small function `run_cmd`, which I then replace with during testing with another function `_wrap_run_cmd`. Both of these are in https://github.com/goerz/clusterjob/blob/develop/clusterjob/utils.py. The test is in https://github.com/goerz/clusterjob/blob/develop/tests/test_workflow.py.

natezb, 2016-03-16:
Thanks for the suggestions, I like the idea of doing record/replay for the driver tests. The mocking system will be tricky to implement because the interfaces are very different from driver to driver, but I'm slowly working out an approach.