mabuchilab/Instrumental #12: optics.beam_tools.get_profiles

davidbalslevclausen, 2016-03-07:
I have a Python(X,Y) 2.7.10 installation and have I have installed instrumental-lib using pip on a Windows7 machine. I get the error message below when I try to run the quick-start example from readthedocs.:

Traceback (most recent call last):
  File "C:\Users\....\Instrumental.py", line 18, in <module>
    plot_profile(qt_r, qs_r, '1064nm', cavity_elems, cyclical=True)
  File "C:\Python27\lib\site-packages\instrumental\optics\beam_plotting.py", line 85, in plot_profile
    clipping, zeroat)
  File "C:\Python27\lib\site-packages\instrumental\optics\beam_tools.py", line 133, in get_profiles
    z = linspace(cur_z, cur_z + el.d, 10000, endpoint=(i == len(rev_elems)-1))
  File "C:\Python27\lib\site-packages\numpy\core\function_base.py", line 93, in linspace
    dt = result_type(start, stop, float(num))
TypeError: data type not understood

davidbalslevclausen, 2016-03-07:
fixable by making the following changes in beam_tools.py function get_profiles line 131:

        # Get beam profile inside 'Space' elements
        if isinstance(el, Space):
            eld = el.d.to('mm').magnitude
            cur_zd = cur_z.to('mm').magnitude
            z = Q_(linspace(cur_zd, cur_zd + eld, 10000, endpoint=(i == len(rev_elems)-1)),'mm')

natezb, 2016-03-07:
Hi David, thanks for the bug report. What versions of `pint` and `numpy` are you using?

davidbalslevclausen, 2016-03-07:
pint version 0.7.2
numpy version 1.10.4

natezb, 2016-03-08:
It was broken for me too, on pint 0.6 and numpy 1.9.2. I'm not sure exactly what changed since this was written, but I fixed it by writing and using a unit-aware linspace function.

commit 75e1f639 (2016-03-08): Fix use of quantities with np.linspace, fixes #12
commit dcdc77c5 (2016-10-04): Fix use of quantities with np.linspace, fixes #12
closed 2016-03-08 by natezb