mabuchilab/Instrumental #13: Index error with ni.py

jmfine, 2016-04-01:
When running through the basic example at http://instrumental-lib.readthedocs.org/en/latest/ni-daqs.html, I get a list index out of range error when trying to execute DAQ=NIDAQ('Dev2').

I have confirmed my device is listed as 'Dev2'.  

Thoughts?

natezb, 2016-04-02:
Thanks for the report. Could you post the traceback message you get with the error?

jmfine, 2016-04-02:
I'd be happy to when I get back to my office on Monday.  Question: have you used NIDAQMX and your interface in a threaded format, while attempting to display the data locked to a screen refresh?

Justin Fine
Postdoctoral scholar 
Arizona State University

natezb, 2016-04-02:
No, I have never used DAQmx with threading.

natezb, 2016-06-30:
I'm closing this due to inactivity. Feel free to reopen if you're still experiencing the issue.

closed 2016-06-30 by natezb