mabuchilab/Instrumental #15: DAQ analog output data is assigned to a random channel - fixed

physinet, 2016-04-30:
When I tried writing data to multiple output channels via a Task, I noticed that the output data were coming out of the wrong channel! Which channel each set of data was assigned to was random, and only changed upon resetting the program.

I noticed that ni.py, line 177-178 depends on the order of output channels listed. Python doesn't necessarily always keep dictionaries in the same order, so this was effectively jumbling the output channels.
`ao_names = [name for (name, ch) in self.channels.items() if ch.type == 'AO']
 arr = np.concatenate([Q_(data[ao]).to('V').magnitude for ao in ao_names])`

To ensure consistency, I've added a line before 177 that sorts `ao_names` before creating `arr`, with a short comment explaining the sorting function.
`ao_names.sort(key=lambda x: [int(s) if s.isdigit() else s for s in x.split()]) # key = function sorts by number preferentially`

Let me know if there's a better way of doing this!

physinet, 2016-04-30:
This may not have fixed it... looking into it still.

physinet, 2016-04-30:
It looks like DAQmx really cares what order you add channels to an mxtask. The following code can replace 66-71 and will sort the channels by name. I'm sure there's a more clever way to do this, but I couldn't think of one on the fly.
```
channelarg = []
namearg = []
for arg in args:
    if isinstance(arg, Channel):
        channelarg.append(arg)
        namearg.append(arg.name)
    else:
        channel, name = arg
        channelarg.append(channel)
        namearg.append(channel.name)
        
namearg, channelarg = zip(*sorted(zip(namearg, channelarg))) # sort by name
for i in range(len(channelarg)):
    channel = channelarg[i]
    name = namearg[i]
```

natezb, 2016-05-04:
As you suspected, I believe this was due to the `channels` dict not preserving insertion order. This can be simply fixed by using an `OrderedDict` instead.

Note that the nondeterministic ordering could jumble up the writing of output channels just as well as the reading of input channels.

Feel free to reopen this if the fix does not work for you.

commit fcea391c (2016-05-04): Preserve order that channels were added to a Task. Fixes #15
closed 2016-05-04 by natezb