mabuchilab/Instrumental #18: Problems using Instrumental with py2exe, need option to specify custom defaults_en.txt for pint.

andlier, 2016-06-24:
I'm running into issues using py2exe to package an applicaton using instrumental because the pint-module in Instrumental is configured to only use the default config-files (defaults_en.txt). The issue is discribed in further detail here.
http://stackoverflow.com/questions/30769305/python-py2exe-error-after-compile

It would be convenient to have an option in Instrumental to use custom config text file for pint.

Any ideas how I can solve this without forking Instrumental and using the solution given in the Stack overflow-link?

natezb, 2016-06-24:
I don't have any experience with `py2exe`, so I probably won't be of much help. As Paul B answered on Stack Overflow, the best way would be to figure out how to get `py2exe` to include those data files.

Otherwise, you probably will have to modify Instrumental to use a custom units file. I'll think about adding this as a setting in `instrumental.conf`, though this may make it tricky to unpickle Quantities correctly.

What version of Instrumental are you using?

andlier, 2016-06-28:
Thanks for the quick reply. I looked into how to get py2exe to do what I need, but quickly realized that I could just try one of the other python-freeze solutions. I ended up with pyinstaller, worked out of the box with the exact same project, no issues.

I'm on the 0.2.1 version of instrumental, installed with pip.

natezb, 2016-06-28:
Thanks for the update, good to hear you got your issue resolved.

closed 2016-06-28 by natezb