mabuchilab/Instrumental #20: Attocube ECC100 issues

Percentmk24, 2016-09-19:
Hello,

Im trying to use your driver module to create my own raster scan program using the ECC100. However, I am finding it very difficult to set things like the amplitude and frequency as it keeps giving me a dimensionality error? How do you enter in a value to change these quantities as it is not clear at all in the documentation.

Thanks

Andrew

doddgray, 2016-09-19:
Hi Andrew,

Thanks for your question. It is infinitely easier to diagnose software
problems with example code/backtraces, so if you could provide an example
of an entire python script that reproduces your problem and the error you
see I'm pretty sure we could get you up and running. My guess is that your
issue is with pint <https://pint.readthedocs.io/en/0.7.2/> (the units
library instrumental uses) and how it connects to instrumental. At the
beginning of your script you need to import both the driver you'd like to
use and the pint UnitRegistry object "u" and/or the unitful number
constructor function "Q_" from instrumental with lines like

from instrumental.drivers.motion import ecc100
from instrumental import u, Q_

You need to do this instead of instantiating your own pint UnitRegistry
object, since instrumental always instantiates its own and two different
pint unit registries will not work together. By this I mean that instead
using the code

from instrumental.drivers.motion import ecc100
import pint
u = pint.UnitRegistry()

will cause errors. Once you instantiate your ecc100 object (let's say you
called it 'ecc') you can call its methods with unitful inputs like

ecc.set_amplitude(4*u.mV)

or

ecc.set_amplitude(Q_(4,"millivolt"))

Could this be the issue? We really like pint because it ends up saving a
lot of hassle keeping track of units by hand. If you'd like you can read
all about it via the link above. Hopefully the problem isn't something more
subtle. If this doesn't work, please provide an example of the setup and
error in your instrumental script and we'll try to help.

Best,
Dodd

Percentmk24, 2016-09-21:
Hey Dodd,

Thanks so much for your reply! I think the issue is that i didnt call the units properly. I was entering them as:

ecc100.set_frequency(900)

At the moment, I'm just trying to work at how your python module works so I can start programming a raster scan program using this device. I will try what you have suggested and hopefully it works. Anything I should watch out for when using this module?

Cheers

Andrew

natezb, 2016-09-23:
I'm closing this as it seems to be resolved. Please reopen if the issue is unresolved and you have more information.

closed 2016-09-23 by natezb