mabuchilab/Instrumental #24: Imaging at higher frame rates with Thorlabs DCC1545M

rlung, 2017-03-28:
Hi, I'm trying to use this package to image with the Thorlabs DCC1545M camera with at least 30 fps. In order to do that I need to subsample or select an area of interest (AOI) to reduce the resolution. Subsampling is the ideal solution to maintain the full field of view. Unfortunately, I don't see a way to subsample when looking through the code. I saw a method for binning, but that's not supported with this camera. When I tried setting an AOI, I still get the full resolution image returned to me. To test if I can set an AOI, I tried:
```
cam = instrument(list_instruments()[0])
cam.grab_image(top=0, bottom=400, left=0, right=400)
```
Perhaps that's not the correct way to do it? I'd appreciate any help you can offer!

natezb, 2017-03-28:
Hi, thanks for the bug report!

1) What version of Python are you running?
2) What version of Instrumental are you running?

First, the keyword to use is `bot`, not `bottom`. When I tried to use `bottom`, I got an error. I have the same camera, and I just tested it and it works for me when I use `bot`, returning a 400x400 array.

You might be running an older version of Instrumental that either doesn't allow setting AOI, or doesn't tell you if you've used an invalid keyword. I haven't made a release in awhile, so the git version is pretty far ahead of the last release on PyPI.

Subsampling support should be easy to add, maybe I'll take a crack at it today.

natezb, 2017-03-28:
A quick update: I just added subsampling support to `master`. If you have a chance, let me know if it works for you.

I may soon be refactoring this driver to use `NiceLib` under the hood, but this should do in the meantime.

rlung, 2017-03-28:
Ahh yes of course it's a silly mistake like `bot` vs `bottom`. Subsampling would be a much better option for me anyway. I'll have to check it out tomorrow, but I will let you know if it works. I am using the version from pip install so 0.2.1 it looks like, and I'm on python 2.7.12 I believe (not at the computer so not certain on these). I'll get the version from github tomorrow though.

I'm assuming when you start using NiceLib it won't change how I use this package?

natezb, 2017-03-28:
No, ideally it would not change how you do anything. You will have to have `cffi` installed though. It may also ultimately remove the dependency on `pywin32`.

rlung, 2017-03-31:
Subsampling works great! Imaging at around 30 fps now. Thanks!

closed 2017-03-31 by rlung