mabuchilab/Instrumental #25: Not recognizing Thorlabs camera

rlung, 2017-05-25:
I'm having an issue where I can't see the Thorlabs camera with `list_instruments()` on some computers. I have it working on a couple of computers just fine, but on others, it won't recognize the camera at all. On the computers that it doesn't work, I tried reinstalling and matching the driver for the Thorlab camera to the driver on the working computers. Same OS (Windows 10) on some computers even. Any ideas what the issue could be?

rogers132, 2017-05-25:
A couple of questions:

On the computers that won't recognize the camera, can you recognize/open the camera in the Thorcam software and/or does it show up in device manager as a usb device?  (I have had issues with the usb connection being finicky with these cameras). 
Also on the computers that won't recognize the camera, what happens when you try to import the relevant driver using `from instrumental.drivers.cameras import uc480`?

rlung, 2017-05-25:
Yes, I can get it working on Thorcam software just fine.

I tried `from instrumental.drivers.cameras import uc480`, and I got an error that nicelib is not installed. I installed it with `pip install nicelib`, and when I try `from instrumental.drivers.cameras import uc480` again, I get this error:

```
Module _uc480lib does not yet exist, building it now. This may take a minute...
Searching for headers...
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\Jose\Anaconda2\lib\site-packages\instrumental_lib-0.3.dev1-py2.7.egg\instrumental\drivers\cameras\uc480
.py", line 26, in <module>
    info = load_lib('uc480', __package__)
  File "C:\Users\Jose\Anaconda2\lib\site-packages\nicelib\__init__.py", line 81, in load_lib
    lib_module = _load_or_build_lib(name, pkg)
  File "C:\Users\Jose\Anaconda2\lib\site-packages\nicelib\__init__.py", line 67, in _load_or_build_lib
    build_module.build()
  File "C:\Users\Jose\Anaconda2\lib\site-packages\instrumental_lib-0.3.dev1-py2.7.egg\instrumental\drivers\cameras\_buil
d_uc480.py", line 24, in build
    ignore_system_headers=True)
  File "C:\Users\Jose\Anaconda2\lib\site-packages\nicelib\build.py", line 93, in build_lib
    header_paths, predef_path = handle_header_path(header_info)
  File "C:\Users\Jose\Anaconda2\lib\site-packages\nicelib\util.py", line 58, in handle_header_path
    headers = [find_header(h, include_dirs) for h in header_names]
  File "C:\Users\Jose\Anaconda2\lib\site-packages\nicelib\util.py", line 98, in find_header
    raise Exception("Cannot find header '{}'".format(header_name))
Exception: Cannot find header 'uc480.h'
```

natezb, 2017-05-25:
As the error suggests, it can't find the appropriate header, so either you don't have it installed, or it's located in a directory that Instrumental isn't checking. Can you try to locate this header file? If you can, let us know its path in the filesystem so I we can add it to the search path.

rlung, 2017-05-25:
There are actually 3 files with that name under the Thorlabs directory:
```
C:\Program Files\Thorlabs\Scientific Imaging\DCx Camera Support\Develop\Include
C:\Program Files\Thorlabs\Scientific Imaging\DCx Camera Support\Develop\Source\uc480Acquire
C:\Program Files\Thorlabs\Scientific Imaging\DCx Camera Support\Develop\Source\uc480Live
```

natezb, 2017-05-25:
I just added some paths that should fix this issue for you, so you should pull from GitHub again, test it out, and let me know if you have any other problems.

I believe that once this path issue is sorted out, you may have another one. There's a path-escaping bug in the release version of NiceLib that (I think) applies to this driver. If you install the current `git` version of NiceLib, that should fix the issue. I'll also try to put out a bugfix release of NiceLib soon so you can just install it via pip.

rlung, 2017-05-25:
So I pulled from GitHub and tried `list_instruments()` again and get this:

```
Module _uc480lib does not yet exist, building it now. This may take a minute...
Searching for headers...
Found ['C:\\Program Files (x86)\\Thorlabs\\Scientific Imaging\\DCx Camera Support\\Develop\\Include\\uc480.h']
Parsing and cleaning headers...
Compiling cffi module...
Writing macros...
Done building _uc480lib
Module _nilib does not yet exist, building it now. This may take a minute...
Searching for headers...
[]
```
Tried it again and just got the last half:
```
Module _nilib does not yet exist, building it now. This may take a minute...
Searching for headers...
[]
```
I then uninstalled NiceLib and installed the one from GitHub. Now I just keep getting this from `list_instruments()`:
```
Module _uc480lib does not yet exist, building it now. This may take a minute...
Searching for headers...
Module _nilib does not yet exist, building it now. This may take a minute...
Searching for headers...
[]
```
Also, still able to use the camera through Thorcam.

natezb, 2017-05-25:
Could you do the explicit `from instrumental.drivers.cameras import uc480` and report back what that gives you?

rlung, 2017-05-25:
Looks like the same error from before.
```
Module _uc480lib does not yet exist, building it now. This may take a minute...
Searching for headers...
Traceback (most recent call last):
 File "<stdin>", line 1, in <module>
 File "C:\Users\Jose\Anaconda2\lib\site-packages\instrumental_lib-0.3.dev1-py2.7.egg\instrumental\drivers\cameras\uc480
.py", line 26, in <module>
   info = load_lib('uc480', __package__)
 File "C:\Users\Jose\Anaconda2\lib\site-packages\nicelib\__init__.py", line 81, in load_lib
   lib_module = _load_or_build_lib(name, pkg)
 File "C:\Users\Jose\Anaconda2\lib\site-packages\nicelib\__init__.py", line 67, in _load_or_build_lib
   build_module.build()
 File "C:\Users\Jose\Anaconda2\lib\site-packages\instrumental_lib-0.3.dev1-py2.7.egg\instrumental\drivers\cameras\_buil
d_uc480.py", line 24, in build
   ignore_system_headers=True)
 File "C:\Users\Jose\Anaconda2\lib\site-packages\nicelib\build.py", line 93, in build_lib
   header_paths, predef_path = handle_header_path(header_info)
 File "C:\Users\Jose\Anaconda2\lib\site-packages\nicelib\util.py", line 58, in handle_header_path
   headers = [find_header(h, include_dirs) for h in header_names]
 File "C:\Users\Jose\Anaconda2\lib\site-packages\nicelib\util.py", line 98, in find_header
   raise Exception("Cannot find header '{}'".format(header_name))
Exception: Cannot find header 'uc480.h'
```

natezb, 2017-05-25:
Is this all on the same computer? This seems odd to me, because in your previous comment it appears that when you first called `list_instruments()`, the header was found and the bindings were built. Then you tried `list_instruments()` again, which correctly noticed that the bindings were already built.

Now in your latest comment, the explicit import is no longer finding the existing bindings, nor the header, which doesn't make much sense to me. Is it possible that you have (1) an installed copy of Instrumental and  (2) a different copy from GitHub which is sitting around uninstalled? It seems to me like your most recent comment is still using the older version.

I'll be trying to test this on one of my machines with a fresh copy to see if I have similar issues.

rlung, 2017-05-30:
Yes, all the same computer. Not sure what is going on.

I actually uninstalled Instrumental and made sure Python could not import it. Then I installed the version from GitHub, and I get this error now:
```
C:\Users\Jose\Anaconda2\lib\site-packages\nicelib\nicelib.py:694: UserWarning: No lib function found with a name ending
in 'SetImageSize', with any of these prefixes: ('is_', 'IS_', u'')
  "any of these prefixes: {}".format(name, flags['prefix']))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "instrumental\drivers\cameras\uc480.py", line 228, in <module>
    lib.AOI_IMAGE_SET_POS_FAST_SUPPORTED: 'UINT*',
  File "C:\Users\Jose\Anaconda2\lib\site-packages\nicelib\nicelib.py", line 765, in __getattr__
    value = getattr(self._ffilib, name)
AttributeError: cffi library 'C:\Windows\system32\uc480_64.dll' has no function, constant or global variable named 'AOI_
IMAGE_SET_POS_FAST_SUPPORTED'
```

natezb, 2017-05-30:
Ok, this is a good sign, the bindings have been built and it's importing the meat of the module. How long ago did you pull? There's an update I pushed (c34f4b1) on Friday that should fix this issue, which has to do with changes made in more recent versions of the library. Try pulling and reinstalling once more and you should at least be able to import the module, and with any luck your camera should work as well.

rlung, 2017-05-31:
Okay so I updated with your recent push. Here's what I get with `from instrumental.drivers.cameras import uc480`:
```
C:\Users\Jose\Anaconda2\lib\site-packages\nicelib\nicelib.py:694: UserWarning: No lib function found with a name ending
in 'SetImageSize', with any of these prefixes: ('is_', 'IS_', u'')
  "any of these prefixes: {}".format(name, flags['prefix']))
```
But I can see the camera with `list_instruments`. This is the output:
```
Module _nilib does not yet exist, building it now. This may take a minute...
Searching for headers...
[<UC480_Camera '4102887385'>]
```

natezb, 2017-05-31:
Ok, that's good. The `SetImageSize` warning is due to that function being deprecated and now removed, so I have to update the driver to replace it.

There's also still some noise from NiceLib about `_nilib` which is normal. I'll be updating NiceLib some time soon to disable this kind of output by default.

Feel free to close this if you feel it's been resolved.

rlung, 2017-06-01:
I actually am unable to get the camera working. I am able to find the camera from `list_instruments`, but when I try to open it with `instrument`, I get this:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\Jose\Anaconda2\lib\site-packages\instrumental_lib-0.3.dev1-py2.7.egg\instrumental\drivers\__init__.py",
 line 507, in instrument
    new_inst = mod._instrument(params)
  File "C:\Users\Jose\Anaconda2\lib\site-packages\instrumental_lib-0.3.dev1-py2.7.egg\instrumental\drivers\cameras\uc480
.py", line 327, in _instrument
    return UC480_Camera(**d)
  File "C:\Users\Jose\Anaconda2\lib\site-packages\instrumental_lib-0.3.dev1-py2.7.egg\instrumental\drivers\cameras\uc480
.py", line 489, in __init__
    self._open()
  File "C:\Users\Jose\Anaconda2\lib\site-packages\instrumental_lib-0.3.dev1-py2.7.egg\instrumental\drivers\cameras\uc480
.py", line 551, in _open
    self._allocate_mem_seq(num_bufs)
  File "C:\Users\Jose\Anaconda2\lib\site-packages\instrumental_lib-0.3.dev1-py2.7.egg\instrumental\drivers\cameras\uc480
.py", line 592, in _allocate_mem_seq
    self._dev.SetImageSize(self._width, self._height)
AttributeError: 'Camera' object has no attribute 'SetImageSize'
```

natezb, 2017-06-01:
I replaced the usage of `SetImageSize` in a commit yesterday, so this should be resolved now. I tested it on a separate computer that uses version 4.80 of the Thorlabs library, and it opens and grabs images fine. Let me know if it works for you.

natezb, 2017-06-12:
I'm closing this since the `list_instruments()` aspect of this is fixed. Feel free to open a new issue if you have more problems with this driver.

closed 2017-06-12 by natezb