mabuchilab/Instrumental #26: Fix IDS uEye support in UC480 driver

natezb, 2017-05-30:
After switching to NiceLib for this driver, we're currently only searching for the Thorlabs version of the header (`uc480.h`), not the IDS version (`uEye.h`). I think it's reasonable to implement support for multiple sets of possible headers within NiceLib, just as multiple possible lib binaries are supported.

eedm, 2017-07-18:
I havent tried Instrumental since the addition of NiceLib but after cloning the latest version my script for using the Thorlabs CMOS camera (IDS uEye) fails as follows:

from instrumental.drivers.cameras.uc480 import UC480_Camera,list_instruments

error                                     Traceback (most recent call last)
<ipython-input-3-9ab33a6d8cc8> in <module>()
----> 1 from instrumental.drivers.cameras.uc480 import UC480_Camera,list_instruments

C:\Anaconda3\lib\site-packages\instrumental_lib-0.4.dev0-py3.6.egg\instrumental\drivers\cameras\uc480.py in <module>()
     78 
     79 
---> 80 class NiceUC480(NiceLib):
     81     _info = info
     82     _prefix = ('is_', 'IS_')

C:\Anaconda3\lib\site-packages\nicelib-0.4.dev1-py3.6.egg\nicelib\nicelib.py in __new__(metacls, clsname, bases, classdict)
    606         for name in dir(lib):
    607             try:
--> 608                 attr = getattr(lib, name)
    609                 if ffi and isinstance(attr, ffi.CData) and ffi.typeof(attr).kind != 'function':
    610                     dir_lib.append(name)

error: symbol 'is_PersistentMemory' not found in library 'C:\Windows\system32\uc480_64.dll': error 0x7f

natezb, 2017-07-19:
@eedm thanks for the report! This is an interesting error, because several things appear to be going on. As the error suggests, your DLL is missing `is_PersistentMemory`. However, NiceLib is supposed ignore this issue by catching the `AttributeError`. This isn't working for you because `cffi` is raising an `FFIError` instead (I think this changed between versions of `cffi` at some point). Commit mabuchilab/NiceLib@7ed5fe9a3a1b0880d625afe2745ba18b2d0f0fcc should hopefully fix this for you, so please give it a try if you get a chance.

What's odd is that this means your header does not match the library you have installed. If I recall correctly, you were using the IDS uEye DLLs before, yet I see that NiceLib is accessing `uc480_64.dll`. So it could be that it's using a uEye header with an older version of the uc480 lib. It appears that `is_PersistentMemory` was only added to the API this past January.