mabuchilab/Instrumental #30: cannot import name errortext

benshep, 2017-09-26:
I'm trying to talk to a PCO camera using Instrumental. I don't see any output from `list_instruments()`, instead I get the following:

    Module _uc480lib does not yet exist, building it now. This may take a minute...
    Searching for headers...
    Module _nilib does not yet exist, building it now. This may take a minute...
    Searching for headers...

When I try `import instrumental.drivers.cameras.pco`, I get the following:

    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
      File "instrumental\drivers\cameras\pco.py", line 18, in <module>
        from ._pixelfly import errortext
    ImportError: cannot import name errortext

Any ideas? I'm probably doing something stupid, but I can't quite see what. I've got the latest version from Github, but `errortext` doesn't seem to be present.

natezb, 2017-09-27:
Thanks for your report, and for finding the specific exception. No, you're not doing anything stupid. This error-message module isn't really set up for easy installation yet. Give this a try and see if it works for you:

First, you should make sure you have a C-compiler available for `cffi` to use.  See [here](https://cffi.readthedocs.io/en/latest/installation.html) for info on how to do this for your platform. Then `cd` into your `Instrumental` root directory, then run `python instrumental/drivers/cameras/_pixelfly/_cffi_build/build_errortext.py`, which should build the `errortext` module. If you're running Windows and it complains about not finding `vcvarsall.bat`, you may need to add `import setuptools` at the top of `build_errortext.py`, as noted on that `cffi` page I linked.

Rather than running this script manually, you also may just be able to reinstall Instrumental after your C compiler is set up. `setup.py` is supposed to look for any of these special build files and run them.

How did you install Instrumental: from PyPI, or by downloading and installing from source?

Let me know if that helps, or if you're still having issues. I wish this `errortext` business were simpler, but it's what we're stuck with for now.

benshep, 2017-09-27:
Thanks. I am on Windows, so obviously anything involving compilation is going to be hard (going by bitter experience). I have MS Visual C++ Express; hopefully that should be OK.

I installed from PyPI.

I'll try it again tomorrow and let you know. Thanks for the assistance!

natezb, 2017-09-27:
Yeah, I know the feeling. I'd like to avoid the compilation requirement as much as possible (since I use Instrumental on Windows, too), but the error-code-to-text function from PCO is provided only as source in the header, not pre-compiled in the DLL.

I've considered replacing that function with my own pure-python version to avoid this issue, but was never sure it was worth the effort. I'm still unsure.

zakv, 2019-12-09:
I know this issue is pretty old, but it's still open so I thought I'd post a workaround that I've been using before coming across this discussion. The code below monkey patches a new version of get_error_text() into instrumental.drivers.camers.pco which simply prints out the error's hex code. For convenience it also prints out the hex code with some of the bits masked. If the user simply runs searches for the masked error code in the header PCO_errt.h, they'll be taken to the line with text describing the error.

Admittedly this is missing some nice features. For example, it doesn't interpret the error's layer code and it requires the user to manually look through PCO_errt.h, so it's not as nice as the compiled GetErrorText() function.

As mentioned above, replacing GetErrorText() with a pure python function would be tedious and avoiding compilation is desirable. One workaround might be to modify instrumental's get_error_text() to call the compiled GetErrorText() if it is available, and fall back to the method below if the compiled code isn't available. Maybe this would be an reasonable compromise?

```
# Monkey patch inst.drivers.cameras.pco.get_error_text()
from instrumental.drivers.util import as_enum

def get_error_text_monkey_patch(ret_code):
    # Get hex version of return code for output string
    ret_code_hex = ret_code.to_bytes(4, byteorder="big", signed=True).hex().upper()
    if ret_code_hex[0]=="A":
        # "Common" errors start with A and have more bits masked in PCO_errt.h
        # In this case we need to mask all but first 4 bits and last 8 bits
        mask = int.from_bytes(b"\xF0\x00\x00\xFF", byteorder="big", signed=True)
    else:
        # For other errors fewer bits are masked
        mask = int.from_bytes(b"\xF0\x00\xFF\xFF", byteorder="big", signed=True)
    
    # This masked code should appear in PCO_errt.h in a comment next to the error string
    masked_ret_code_hex = (ret_code & mask).to_bytes(4, byteorder="big", signed=True).hex().upper()
    
    return "PCO Error code: " + ret_code_hex + ", Look for " + masked_ret_code_hex + " in PCO_errt.h"

inst.drivers.cameras.pco.get_error_text = get_error_text_monkey_patch
```

zakv, 2020-04-10:
A `PCOError` class based on the code in the previous comment was added to Instrumental in PR https://github.com/mabuchilab/Instrumental/pull/105. It calls the compiled `GetErrorText()` function if available, and if it isn't available it instead prints out the error's hex code so that the user can look up its meaning in the PCO SDK's PCO_errt.h header.

That isn't quite as nice as having a pure python error-code-to-error-text function as Nate mentioned, but it's a reasonable workaround. Given that, this issue could potentially be closed, and it could be reopened later if it's decided that a pure python version would be worthwhile.