mabuchilab/Instrumental #37: Thorlabs camera not recognised

anaandresarroyo, 2017-11-23:
Hello,

I am trying to connect to a Thorlabs camera UI-3240LE_NIR-GL but I'm getting errors and the camera is not recognised.

When I try 

`from instrumental.drivers.cameras import uc480`

The output is:

```
Module _uc480lib does not yet exist, building it now. This may take a minute...
Searching for headers...
Traceback (most recent call last):
  File "R:\aa938\NanoPhotonics\Python\camera.py", line 9, in <module>
    from instrumental.drivers.cameras import uc480
  File "c:\users\ana andres\documents\github\instrumental\instrumental\drivers\cameras\uc480.py", line 28, in <module>
    info = load_lib('uc480', __package__)
  File "C:\Program Files\Anaconda2\lib\site-packages\nicelib\__init__.py", line 81, in load_lib
    lib_module = _load_or_build_lib(name, pkg)
  File "C:\Program Files\Anaconda2\lib\site-packages\nicelib\__init__.py", line 67, in _load_or_build_lib
    build_module.build()
  File "c:\users\ana andres\documents\github\instrumental\instrumental\drivers\cameras\_build_uc480.py", line 35, in build
    ignore_system_headers=True)
  File "C:\Program Files\Anaconda2\lib\site-packages\nicelib\build.py", line 93, in build_lib
    header_paths, predef_path = handle_header_path(header_info)
  File "C:\Program Files\Anaconda2\lib\site-packages\nicelib\util.py", line 53, in handle_header_path
    raise KeyError("Header dict must contain key 'header'")
KeyError: "Header dict must contain key 'header'"
```

And when I try

```
from instrumental import list_instruments
paramsets = list_instruments()
print paramsets
```

The output is empty.

I have checked that the camera can be seen by the ThorCam software.

Thank you for any advice.

natezb, 2017-11-23:
Hi Ana,

Could you tell me what versions of NiceLib and Instrumental you are using? You can find this by doing:

    import nicelib, instrumental
    print(nicelib.__version__)
    print(instrumental.__version__)

My first guess is that you may be using an older release of NiceLib, as the form in which headers are specified has changed a bit in the most recent release.

anaandresarroyo, 2017-11-24:
Hello,

Instrumental is version 0.5.dev1. 
NiceLib was version 0.3.1. 
I uninstalled NiceLib and installed it again so it's now version 0.4

Both `list_instruments()` and `instrument('uc480')` work now.

Thank you very much.

natezb, 2017-11-24:
Awesome, I'm glad it worked.

closed 2017-11-24 by natezb