mabuchilab/Instrumental #42: Cannot import name list_instruments / instrument 'module' object is not callable

anaandresarroyo, 2017-12-14:
Hello,

I have updated my instrumental version from 0.5.dev.1 to 0.5.1 (nicelib is the same at version 0.4) and I now get the following errors:

When I try: `from instrumental import list_instruments`
I get: `ImportError: cannot import name list_instruments`

When I try: 
```
from instrumental import instrument
instrument('uc480')
```
I get: TypeError: `'module' object is not callable`

My first thought was that `list_instruments` and `instrument` have been re-organised but I I haven't been able to find their new location, and I'm not sure if I have inadvertently changed something else.

I updated the instrumental library in my computer because, even thought it was working well, I tried installing it in another computer to control a UI-3240LE camera and it was giving me these errors. Therefore I wanted to have both computers with the latest version of instrumental and then my computer started giving these errors too.

Thank you for any advice.

natezb, 2017-12-14:
Unfortunately, there is another (unrelated) package on PyPI that is named `instrumental`. Since we haven't released 0.5 yet (the latest release is 0.4.2), I am assuming that you inadvertently installed that package. On PyPI, our package is called `instrumental-lib`.

I hope that helps, and please let me know if there are any more issues.

anaandresarroyo, 2017-12-14:
Yes, that seems to have been the case. I uninstalled the other instrumental package and then  I cloned this GitHub repository and installed it by running the setup.py file. It now says it's version 0.5.dev1 and everything is working again. Thank you for the quick reply.

natezb, 2017-12-14:
Great, good to hear it's working again.

closed 2017-12-14 by natezb