mabuchilab/Instrumental #43: list_instruments() does not see Thorlabs DCx camera

Noahyt, 2017-12-18:
I am using python 3.6.1 and instrumental 0.4.2.

As per the advice in [this](https://github.com/clinton-hall/nzbToMedia/issues/477) issue, I ran the import command  `from instrumental.drivers.cameras import uc480`.

The error I got was: 

>
>  Traceback (most recent call last):
>   File "<input>", line 1, in <module>
>   File "C:\Program Files\JetBrains\PyCharm Community Edition 2017.3\helpers\pydev\_pydev_bundle\pydev_import_hook.py", line 20, in do_import
>     module = self._system_import(name, *args, **kwargs)
>   File "C:\Users\ntoyonaga\AppData\Local\Continuum\anaconda3\envs\test_env\lib\site-packages\instrumental\drivers\cameras\uc480.py", line 15, in <module>
>     import win32event
>   File "C:\Program Files\JetBrains\PyCharm Community Edition 2017.3\helpers\pydev\_pydev_bundle\pydev_import_hook.py", line 20, in do_import
>     module = self._system_import(name, *args, **kwargs)
> ModuleNotFoundError: No module named 'win32event'
> 

Any help much appreciated!

Noahyt, 2017-12-18:
After some searching I realized I just needed to install a packages (pypiwin32 for those with the same error).  However, I now have a new error when I try to run the `list_instruments()` command: 

> AttributeError: module 'instrumental.drivers' has no attribute 'saved_instruments'

This is strange to me as before the command ran fine but did not see the camera, and now it appears not to be working at all...

natezb, 2017-12-18:
Thanks for the report. This does appear to be an issue that was fixed in 33287b04567e278a888cde1b1d6817566b49205e (which hasn't yet hit a release).

It is odd to me that the issue manifested itself only after you installed pypiwin32. Could you post your code and a full stack trace from the error?

In the meantime, you might consider updating to the current dev version on GitHub, or might be able to import `list_instruments` . from `instrumental.drivers`.

natezb, 2017-12-18:
After doing a fresh install of 0.4.2, I get the same issue when I run:

    >>> import instrumental
    >>> instrumental.list_instruments()

However, this works:

    >>> from instrumental import list_instruments
    >>> list_instruments()
    []

This odd behavior is due to the lazy loading in the toplevel package. So, if you're just interested in getting things to work for now (and can wait for the real fix down the road), I'd recommend running `list_instruments` this way.

natezb, 2017-12-18:
Added a test for catching this sort of issue in 70817881f4ed0897fc24ec00cb6c182482fdbd2d and 7f01079fd64e285c149d3a28bab2b3d28a4665eb.

Noahyt, 2017-12-18:
@natezb Thanks! The quick fix you indicated works for me.

closed 2017-12-18 by Noahyt