mabuchilab/Instrumental #44: Cannot Load Thorlabs Camera Parameters from file

Noahyt, 2017-12-21:
I am using a Thorlabs DCx camera and trying to use the load_params() command.  I have written a .tcp file from ThorCam and want to load it during use with Instrumental. 

The error I am getting is: 

> Traceback (most recent call last):
>   File "C:/Users/ntoyonaga/Documents/test/multiCamera.py", line 20, in <module>
>     cam1.load_params('camProfile.tcp')
>   File "C:\Users\ntoyonaga\AppData\Local\Continuum\anaconda3\envs\test_env\lib\site-packages\instrumental\drivers\cameras\uc480.py", line 488, in load_params
>     self._dev.ParameterSet(cmd, param, ffi.sizeof(param))
>   File "C:\Users\ntoyonaga\AppData\Local\Continuum\anaconda3\envs\test_env\lib\site-packages\nicelib\nicelib.py", line 824, in __call__
>     result = self._func(*(self._handles + args), niceobj=self._niceobj)
>   File "C:\Users\ntoyonaga\AppData\Local\Continuum\anaconda3\envs\test_env\lib\site-packages\nicelib\nicelib.py", line 433, in wrapped
>     "{}".format(inarg, argtype))
> TypeError: Cannot convert <cdata 'wchar_t[]' owning 30 bytes> to required type<ctype 'void *'>

natezb, 2017-12-23:
Hey, sorry for the delay. I'm away from the lab until early January, so I may not be the most helpful in the meantime.

I think this is a bug in `nicelib`. From looking at it briefly, it appears that `'inout'` arguments don't get converted in the same way that `'in'` arguments do (with `_wrap_inarg()`), so the `wchar_t[]` doesn't get auto-converted to `void *`. It's possible that all that's needed is to replace the `arg = ffi.new(argtype, inarg)` with `arg = _wrap_inarg(ffi, argtype, inarg)`. I'd have to think a bit about the semantics of `'in'` vs `'inout'` before adding this change, but feel free to try this out and see if it fixes your problem.

Noahyt, 2018-01-05:
@natezb I was also out of the lab until a few days ago, so no worries.  

Apologies if this is an obvious question -- but where should I be performing the line replacement you suggest?  In uc480.py?

Thanks!

natezb, 2018-01-06:
The substitution would be in `nicelib`'s `nicelib.py`. Now that I'm back, I can probably take a quick look at this on Monday and try to figure out the best way to handle `inout` args.

natezb, 2018-01-08:
I added change mabuchilab/nicelib@bd67b2c761a126f0b9ba29ff1bb5037f13f44078 that fixes the casting issue, so you can install that from GitHub if you're interested.

I also noticed (and fixed) a few latent issues in `load_params()`. I don't think I've used that function in years and it probably hasn't been tested (until now) since switching to using nicelib. Give this a try if you get a chance and let me know if it's working for you.

Noahyt, 2018-01-09:
Using your updated version of nicelib, I get a new error:

>  Traceback (most recent call last):
>   File "C:/Users/ntoyonaga/Documents/test/multiCamera.py", line 21, in <module>
>     cam1.load_params("camProfile.tcp")
>   File "C:\Users\ntoyonaga\AppData\Local\Continuum\anaconda3\envs\test_env\lib\site-packages\instrumental\drivers\cameras\uc480.py", line 505, in load_params
>     depth = depth_map[mode]
> KeyError: 1

natezb, 2018-01-09:
I should have mentioned that you have to install the current git version of instrumental, too, to get the fixes to `load_params()` that I made yesterday in 9d72e47, which should fix the issue you're encountering.