mabuchilab/Instrumental #45: Cannot set the integration time for CCS175 Thorlabs Spectrometer

hahaits, 2017-12-29:
[image]

natezb, 2017-12-29:
This means you should pass a `pint` `Quantity` (or a sting like `'50 ms'`) for the `integration_time`.

We should fix the docstring of `take_data()` to indicate this. We should also consider adding additional detail to the error message to make it less confusing for those unfamiliar with `pint`.

I'll keep this open until those items have been addressed, but please let me know if using units fixes your problem, or if you continue to have issues.

hahaits, 2018-01-05:
Hi,

a string like '50 ms'

natezb, 2018-01-06:
Hey, did you mean to say more than this? Or did you accidentally reply?

hahaits, 2018-01-08:
Sorry, I meant to say, a string like ‘50 ms‘ worked, I used Pint as well but that didn’t work.

Hopefully the info comes in handy to make any changes to the library. If you do not remember, I had an issue with the integration time for Thorlabs CCS175.

closed 2018-01-10 by natezb