mabuchilab/Instrumental #46: Issue when using multiple thorlabs 480x cameras

Noahyt, 2018-01-05:
This problem is the underlying reason I made -- but rather than self-diagnosing perhaps this is a more general issue in Instrumental.  

I have loaded two thorlabs camera objects, cam1 and cam2.  They are physically identical and I am using them in a stereo application so they are under similar physical lighting conditions.  When I use the "ThorCam" application they obviously produce similar quality images.  

However, when I try to load them and take images in instrumental the first image captured (here cam1) always comes out signficantly brighter than the other.  My code is as follows:

```
cam1 = instrument(instrumentList[1])
cam2 = instrument(instrumentList[0])

img1 = cam1.grab_image(exposure_time = "10ms")  ##set exposure using "x_ms"
img2 = cam2.grab_image(exposure_time = "10ms" )
```

I have tried taking multiple images with each camera (since I know opencv has an issue where sometimes repeated image captures are necessary to get the real image.  However, all images taken by cam1 are bright and all by cam2 are dark.  

Additionally, switching the _physical_ camera associated with cam1 and cam2 does not change which is brighter and darker.  In other words whichever physical device is loaded as cam1 and whichever is loaded as cam2, cam1 is brighter and cam2 is darker.  

However, _reversing the order_ (taking cam2 before cam1) does change which is brighter.  When I capture Cam2 first and Cam1 2nd, Cam2 has a bright image and Cam1 a dark image.

Thanks!

natezb, 2018-01-06:
Normally when people post issues I have a pretty good idea of what may be going on, but this one's not so obvious on its face. Given what you've said here (thanks for investigating btw), it's probably a bug somewhere in the `uc480` driver. I'll try to see if I can reproduce this when I'm back in the lab on Monday.

natezb, 2018-01-09:
I have been able to reproduce the issue.

    >>> cam1 = instrument(uc480_id=1)
    >>> cam2 = instrument(uc480_id=2)
    >>> cam1.grab_image().mean(), cam2.grab_image().mean()
    (254.99998397827147, 85.624255371093753)

However, `cam1` behaves normally if you grab an image before opening `cam2`:

    >>> cam1 = instrument(uc480_id=1)
    >>> cam1.grab_image().mean()
     92.610065460205078
    >>> cam2 = instrument(uc480_id=2)
    >>> cam1.grab_image().mean(), cam2.grab_image().mean()
    (97.560542297363284, 86.977066040039063)

Using the newly-fixed `load_params()` does seem to fix this:

    >>> cam1 = instrument(uc480_id=1)
    >>> cam2 = instrument(uc480_id=2)
    >>> cam1.grab_image().mean(), cam2.grab_image().mean()
    (254.99999771118163, 87.158396911621097)
    >>> cam1.load_params(u'path/to/profile.tcp')
    >>> cam2.load_params(u'path/to/profile.tcp')
    >>> cam1.grab_image().mean(), cam2.grab_image().mean()
    (96.078477478027338, 87.544498443603516)

Interestingly, running `ResetToDefault` does not fix the issue.

Noahyt, 2018-01-09:
Hmm -- I didn't think of grabbing an image in between loading the cameras, but this does fix my issue. 

I'd like to use the load_parameters(), but that is still not working for me.  However I think I can close this issue since this appears to be purely related to #44 .

natezb, 2018-01-09:
I'm going to reopen this, since we only have a workaround, not a fix. This is certainly unexpected behavior that should be fixed at some point if we can figure out what's going on.

doddgray, 2018-01-09:
I'm not sure how relevant this is, but before this issue is forgotten I wanted to mention that I have noticed serious issues when trying to simultaneously capture video from a Thorlabs 480x CMOS camera and spectra from Thorlabs CCS175 spectrometer (which is based on a CCD line-scan camera) using Thorlabs' own software (Thorcam and Thorlabs OSA, respectively). Specifically, if I try to capture image frames from the camera while the spectrometer is already running, the spectrometer stops working and vice versa. This has nothing to do with Instrumental or even Python, but I think there's a chance that a similar issue is to blame for what you're seeing with two 480x cameras here. I imagine that these two instruments are being read-out using very similar drivers, and it is possible that the Thorlabs drivers were not written to handle simultaneous read-out very well (as evidenced by my experience with the spectrometer and camera). To check whether this is an Instrumental issue or a driver issue, maybe try reading frames from one camera with Thorcam while accessing the other the Python/Instrumental?