mabuchilab/Instrumental #48: Cannot connect to Bristol 721 Spectrometer

fsponciano, 2018-03-13:
I've tried to connect to a Bristol 721 spectrometer (NIR model), but there appears to be an error in the initial communication to initialise the device. It appears to be a checksum error, and the command line continues displaying this message until a keyboard interrupt is sent. Any ideas as to why this is happening?

natezb, 2018-03-13:
Thanks for the report. Could you copy the output that you're seeing and provide it here?

fsponciano, 2018-03-13:
[Instrumental_bristolError.txt](https://github.com/mabuchilab/Instrumental/files/1807905/Instrumental_bristolError.txt)

fsponciano, 2018-03-13:
I did have another program running, just as a sanity check to see the Bristol was actually recognized by the computer. I've checked in the Windows Device Manager and NI-Max to see the Bristol comes up under serial port COM2. This doesn't seem to be the issue.

I'll go ahead an try using the driver from instrumental directly, see if I can pinpoint the problem down.