mabuchilab/Instrumental #49: Issues connecting with Thorlabs DCC3260C Camera

berto10, 2018-03-16:
I'm trying to run a thorlabs DCC3260C Camera, but it is not showing in the instruments list. The camera works well with the Thorcams software. Also, I trying using "from instrumental import instrument, list_instruments but get the second error below about not fining library. Any help is appreciated

C:\Users\berto_perez-reyes>python
Python 2.7.13 (v2.7.13:a06454b1afa1, Dec 17 2016, 20:42:59) [MSC v.1500 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from instrumental import instrument, list_instruments
>>> paramsets = list_instruments()
>>> paramsets
[]
>>>

>>> from instrumental.drivers.cameras import uc480
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python27\lib\site-packages\instrumental\drivers\cameras\uc480.py", line 28, in <module>
    info = load_lib('uc480', __package__)
  File "C:\Python27\lib\site-packages\nicelib\__init__.py", line 67, in load_lib
    build_module.build(**kwargs)
  File "C:\Python27\lib\site-packages\instrumental\drivers\cameras\_build_uc480.py", line 35, in build
    ignore_system_headers=True)
  File "C:\Python27\lib\site-packages\nicelib\build.py", line 113, in build_lib
    lib_name = handle_lib_name(lib_name, filedir)
  File "C:\Python27\lib\site-packages\nicelib\util.py", line 152, in handle_lib_name
    raise ValueError("Cannot find library '{}'".format(lib_names))
ValueError: Cannot find library '('uc480', 'ueye_api')'
>>>

rogers132, 2018-03-16:
Hi,

Have you added the folder containing the uc480 libraries to your PATH?
More details here:
http://instrumental-lib.readthedocs.io/en/stable/uc480-cameras.html

File "", line 1, in

berto10, 2018-03-16:
I am able to see my camera now. However, following up on this issue. Is there an example on how to capture an image? I tried using "cam.capture()" but get AttributeError: "UC480_Camera object has no attribute 'capture'" Is there example code on how to capture an image with thorlabs DCC3260 camera? Thanks! 

C:\Users\berto_perez-reyes>PYTHON
Python 3.6.3 |Anaconda, Inc.| (default, Oct 15 2017, 03:27:45) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from instrumental import instrument, list_instruments
>>> paramsets=list_instruments()
>>> paramsets
[<ParamSet[UC480_Camera] serial=b'4103261793' model=b'SC1936G11C' id=1>]
>>> cam=instrument(paramsets[0])
>>> cam.capture()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'UC480_Camera' object has no attribute 'capture'
>>>

natezb, 2018-03-16:
There is API documentation on the page Chris linked you to: http://instrumental-lib.readthedocs.io/en/stable/uc480-cameras.html. The simplest method to use for capturing an image is `grab_image()`.

berto10, 2018-03-19:
Thanks! Using grab_image( ) is working just fine for what I'm currently doing! 

Thanks for all your help!

closed 2018-03-19 by berto10