mabuchilab/Instrumental #50: Cannot detect uc480 camera

anaandresarroyo, 2018-03-22:
Hello,

I am trying to use instrumental on another computer to control a uc480 camera but it doesn't find it. The camera works well with instrumental on two other computers, and is accessible via ThorCam on this computer but still I can't find it with python. 

I've got the latest version of instrumental (0.5) and nicelib (0.6.dev0) that I've cloned from GitHub. I've also upgraded the packages pycparser and pywin32. I've tried placing the uc480_64.dll file in the Windows/System32 folder, as well as adding the Thorlabs folder with the dll to the python PATH. `list_instruments()` returns blank, `from instrumental.drivers.cameras import uc480` gives no errors, but `camera = instrument('uc480') says "Exception: Instrument with alias 'uc480' not found in config file".

I suspect something is configured incorrectly but I haven't been able to figure out what. Do you have any suggestions?

Thank you,
Ana

natezb, 2018-03-22:
When  you ran `camera = instrument('uc480')`, it looked in your config file for a saved instrument named 'uc480', hence the error.

You might try listing instruments directly from the `uc480` module, with logging enabled, to get some extra info:
```
import logging
logging.basicConfig(level=logging.DEBUG)

from instrumental.drivers.cameras import uc480
uc480.list_instruments()
```
See if that gives you any clues, and post the log output here if you're still stuck.

anaandresarroyo, 2018-03-29:
Thank you for the reply and the suggestions. I haven't been able to fix it yet but I am even more convinced that it has to do with the python installation/setup on the computer. The logging didn't output much information and I get different errors when running the same script from different file locations.

closed 2018-03-29 by anaandresarroyo