mabuchilab/Instrumental #51: Error UC480: AttributeError: 'Sig' object has no attribute 'args'

Gene2002, 2018-03-24:
Hello, I have been attempting to set up instrumental for UC308, but I have confronted this error message. I am unsure of what the problem is and I have tried to go through the files and search for a possible cause. Am I lacking a dependency or am I incorrectly using the class?

This error occurs whenever start_capture() is called. Thank you!

natezb, 2018-03-24:
Thanks for the report. Could yo u post the full traceback of the error you get?

Gene2002, 2018-03-25:
>>> cam.start_capture()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python27\lib\site-packages\instrumental\drivers\cameras\uc480.py", line 753, in start_capture
    self._dev.CaptureVideo(lib.DONT_WAIT)  # Trigger
  File "C:\Python27\lib\site-packages\nicelib\nicelib.py", line 677, in __call__
    return self._libfunc(*(self._niceobj._handles + args), niceobj=self._niceobj)
  File "C:\Python27\lib\site-packages\nicelib\nicelib.py", line 1126, in __call__
    return self.sig.extract_outputs(c_args, retval, ret_handler_args)
  File "C:\Python27\lib\site-packages\nicelib\nicelib.py", line 198, in extract_outputs
    retval = self.ret_handler.handle(retval, ret_handler_kwargs)
  File "C:\Python27\lib\site-packages\nicelib\nicelib.py", line 521, in handle
    return self.__func__(retval, **kwargs)
  File "C:\Python27\lib\site-packages\instrumental\drivers\cameras\uc480.py", line 78, in wrap
    err_code, err_msg = niceobj.GetError()
  File "C:\Python27\lib\site-packages\nicelib\nicelib.py", line 677, in __call__
    return self._libfunc(*(self._niceobj._handles + args), niceobj=self._niceobj)
  File "C:\Python27\lib\site-packages\nicelib\nicelib.py", line 1126, in __call__
    return self.sig.extract_outputs(c_args, retval, ret_handler_args)
  File "C:\Python27\lib\site-packages\nicelib\nicelib.py", line 195, in extract_outputs
    if handler.makes_output]
  File "C:\Python27\lib\site-packages\nicelib\nicelib.py", line 490, in extract_output
    free_buf = self.sig.args['free_buf']
AttributeError: 'Sig' object has no attribute 'args'

natezb, 2018-03-26:
It appears to be a bug in the wrapping of `GetError`, which is being called due to an error code returned by `CaptureVideo`. The problem is a simple typo in nicelib: That `self.sig.args['free_buf']` in the traceback should instead be `self.sig.flags['free_buf']`. I'll fix this in nicelib and hopefully push out a hotfix soon. In the meantime, you can make the fix locally and see if it fixes the problem.

Note that `CaptureVideo` is returning an error, so you probably have some other error to work through as well.

natezb, 2018-07-10:
This should be fixed in the now-released NiceLib 0.5.2

closed 2018-07-10 by natezb