mabuchilab/Instrumental #52: Thorlabs Camera Viewing Live Image

berto10, 2018-05-07:
Hello, 

I have a Thorlabs camera that I'm trying to be able to get live images from. I have gotten my camera to work with xxxx.grab_image(   ) but I'm not able to make it work with xxxx.start_live_video(   ).

I also tried running camera_gui.py and it would not work. I got the following error. 

"
image = QImage(arr.data, self.camera.width, self.camera.height, bpl, format)

TypeError: arguments did not match any overloaded call:
  QImage(): too many arguments
  QImage(QSize, QImage.Format): argument 1 has unexpected type 'memoryview'
  QImage(int, int, QImage.Format): argument 1 has unexpected type 'memoryview'
  QImage(str, int, int, QImage.Format): argument 1 has unexpected type 'memoryview'
  QImage(sip.voidptr, int, int, QImage.Format): argument 1 has unexpected type 'memoryview' 

"

Any advice is greatly appreciated! 

Thanks!

natezb, 2018-05-08:
Thanks for the error report. What versions of Instrumental, python, and pyqt are you using?

berto10, 2018-05-08:
The version that I am using are the following: 
Instrumental 0.4.2
python 3.6.3
pyqtZ 0.1.2

natezb, 2018-05-09:
I'm not sure what pyqtZ is, but the number you gave doesn't appear to be a reasonable recent pyqt version number. You can check it with something like the following (for PyQt5):
```
from PyQt5.QtCore import QT_VERSION_STR
print(QT_VERSION_STR)
```
If you're instead using pyside, the following works:
```
import PySide
print(PySide.__version__)
```
Does the "Grab Frame" button in the camera GUI work for you? If not, the issue when using the GUI may be strictly Qt-related.

As for the live video issue in general, it may help if you posted your code that tried to use `start_live_video()`, along with any error tracebacks you get.

berto10, 2018-05-15:
Below is the code I am using. This is the code that I found in the example section. When I try to look at the list of modules with "pip list" PyQt5 does not show up. However, If I'm in python I'm able to see the version of PyQt5 using the code you described. Thanks for your help! 

Python 3.6.3 |Anaconda custom (64-bit)| (default, Oct 15 2017, 03:27:45) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from PyQt5.QtCore import QT_VERSION_STR
>>> print(QT_VERSION_STR)
5.6.2
>>>

Code for live images: 

`import sys
from qtpy.QtWidgets import (QApplication, QMainWindow, QWidget, QScrollArea, QPushButton,
                            QVBoxLayout, QHBoxLayout)
from instrumental import instrument, gui

def create_window():
    # Create app and widgets
    app = QApplication(sys.argv)
    win = QMainWindow()
    main_area = QWidget()
    button_area = QWidget()
    scroll_area = QScrollArea()
    button = QPushButton("Start Video")
    btn_grab = QPushButton("Grab Frame")

    # Create layouts
    vbox = QVBoxLayout()
    hbox = QHBoxLayout()

    # Fill Layouts
    vbox.addWidget(scroll_area)
    vbox.addWidget(button_area)
    hbox.addStretch()
    hbox.addWidget(button)
    hbox.addWidget(btn_grab)

    # Assign layouts to widgets
    main_area.setLayout(vbox)
    button_area.setLayout(hbox)
    scroll_area.setLayout(QVBoxLayout())

    # Attach some child widgets directly
    win.setCentralWidget(main_area)

    return app, win, button, btn_grab, scroll_area

if __name__ == '__main__':
    cam = instrument(serial=b'4002784060')  # Replace with your camera's alias

    with cam:
        app, win, ssbutton, btn_grab, scroll_area = create_window()
        camview = gui.CameraView(cam)
        scroll_area.setWidget(camview)

        ssbutton.running = False
        def start_stop():
            if not ssbutton.running:
                camview.start_video()
                ssbutton.setText("Stop Video")
                ssbutton.running = True
            else:
                camview.stop_video()
                ssbutton.setText("Start Video")
                ssbutton.running = False
        ssbutton.clicked.connect(start_stop)

        def grab():
            if not ssbutton.running:
                camview.grab_image()
        btn_grab.clicked.connect(grab)

        win.show()
        app.exec_()

        if ssbutton.running:
            camview.stop_video()`

berto10, 2018-05-15:
This is the error I get: 

  File "C:\Users\berto_perez-reyes\AppData\Local\Continuum\anaconda3\lib\site-packages\instrumental\gui.py", line 102, in _wait_for_frame
    self._set_pixmap_from_array(arr)

  File "C:\Users\berto_perez-reyes\AppData\Local\Continuum\anaconda3\lib\site-packages\instrumental\gui.py", line 76, in _set_pixmap_from_array
    image = QImage(arr.data, self.camera.width, self.camera.height, bpl, format)

TypeError: arguments did not match any overloaded call:
  QImage(): too many arguments
  QImage(QSize, QImage.Format): argument 1 has unexpected type 'memoryview'
  QImage(int, int, QImage.Format): argument 1 has unexpected type 'memoryview'
  QImage(str, int, int, QImage.Format): argument 1 has unexpected type 'memoryview'
  QImage(sip.voidptr, int, int, QImage.Format): argument 1 has unexpected type 'memoryview'
  QImage(str, int, int, int, QImage.Format): argument 1 has unexpected type 'memoryview'
  QImage(sip.voidptr, int, int, int, QImage.Format): argument 1 has unexpected type 'memoryview'
  QImage(List[str]): argument 1 has unexpected type 'memoryview'
  QImage(str, format: str = None): argument 1 has unexpected type 'memoryview'
  QImage(QImage): argument 1 has unexpected type 'memoryview'
  QImage(Any): too many arguments

Traceback (most recent call last):

closed 2018-05-20 by natezb