mabuchilab/Instrumental #55: ValueError: Cannot find library '('uc480' , 'ueye_api')'

danielmtzz, 2018-07-12:
I'm getting an error when I try to import:
from instrumental.drivers.cameras import uc480
ValueError: Cannot find library '('uc480' , 'ueye_api')'
I added the correct folders to my PATH, but it's still not working. The camera works just fine with ThorCam.
Windows 64 bit. Can you please help?

natezb, 2018-07-12:
I have a couple of ideas:

1. Have you restarted your terminal since updating the PATH? Verify that the PATH change took effect for this terminal you're using.

2. You say you're on a 64-bit system, but Instrumental is looking for the 32-bit version of these libs. Is it possible you're using 32-bit python, but only have the 64-bit uc480 libraries? If so, you could fix this by using 64-bit python or installing the 32-bit libraries.

danielmtzz, 2018-07-12:
Thanks for your prompt reply. Does it work on both python 2 and 3?

natezb, 2018-07-12:
Yes, it should; it's possible there are latent bugs, though. I personally use it with Python 3 these days.

danielmtzz, 2018-07-12:
I have a mess with installations of both python 2 and python 3. I also realized that the version that I had was 32 bit. I friend of mind just got it working with the 64 bit version of python. As soon as I fix my installations, I think it should work (with the 64 bit). I will let you know.

danielmtzz, 2018-07-12:
Now I'm getting (with Python 3 64bit):
ModuleNotFoundError: No module named 'instrumental.drivers.cameras._uc480lib'

My friend does see this module with python 2

danielmtzz, 2018-07-12:
Ok, so I installed python 2 and it now loads
from instrumental.drivers.cameras import uc480   just fine. The problem is that list_instruments() is now returning an empty array....

natezb, 2018-07-12:
Try enabling logging and running uc480's `list_instruments()` directly to pinpoint the issue:

```
import logging
logging.basicConfig(level=logging.DEBUG)

from instrumental.drivers.cameras import uc480
uc480.list_instruments()
```

natezb, 2018-07-12:
You can post the log here if you need to. Also, if you post another exception, it'll be most useful with the full backtrace so we can see exactly where it came from.

danielmtzz, 2018-07-12:
Nate,

It's still printing an empty array. Where is the logging saved?

natezb, 2018-07-12:
When you enable logging as shown above, the logs will be printed on the terminal. If you want to save to a file instead, you should be able to use the `filename` argument of `logging.basicConfig()`.

danielmtzz, 2018-07-12:
There is nothing being printed in the command line. Also, I noticed that if I remove the PATH for the dll and  run from instrumental.drivers.cameras import uc480 after that, it still seems to work, but the array is still empty when I list instruments. I don't see any difference between having the dll on PATH or not.

danielmtzz, 2018-07-12:
"[ INFO]2018-07-12 23:22:48, 125 nicelib.nicelib: Making C arg for <OutArgHandler>"

danielmtzz, 2018-07-12:
`[ INFO]2018-07-12 23:22:48, 125 nicelib.nicelib: Making C arg for <OutArgHandler>`

natezb, 2018-07-12:
Did your output get truncated somehow? There should be *a lot* more log output, like dozens or hundreds of lines.

danielmtzz, 2018-07-12:
I don't know if the output is truncated. Do you see many more lines when you run those commands? What do you think is going on?

natezb, 2018-07-12:
If I start logging before importing `uc480`, I get many lines. However, when I run `uc480.list_instruments()` I do get your output (repeated 3 times). So, it appears there aren't any errors, yet no cameras are seen.

Are you running any other application that might be using the camera when you're trying to do this?

You also might check what

```
from instrumental.drivers.cameras import uc480
uc480.lib.GetNumberOfCameras()
```
returns. This is calling the Thorlabs library more directly.

danielmtzz, 2018-07-13:
When I run the same commands but without the USB cable connected, I get the same results and identical logs. It's still not seeing the camera for some reason. This is the camera that I'm using, by the way:
https://www.thorlabs.com/thorproduct.cfm?partnumber=CS2100M-USB

It works on ThorCam just fine.

~Daniel

natezb, 2018-07-13:
This is one of the Thorlabs "Scientific" cameras, which means it uses the TSI library, not uc480. You'll have to make sure that `tsi_sdk.dll` is on the path.

If this isn't working, post the debug output you get from running:

```
from instrumental.log import log_to_screen, DEBUG
log_to_screen(level=DEBUG)

from instrumental.drivers.cameras import tsi
tsi.list_instruments()
```

danielmtzz, 2018-07-13:
I get no log output and still printing an empty array when I run those lines you told me. tsi_sdk.dll was already on path.

natezb, 2018-07-13:
I'm not sure what more to tell you at this point; you'll probably have to do some deeper digging yourself. I'd recommend importing `instrumental.drivers.cameras.tsi` and playing around with the functions and objects it defines, seeing if anything makes sense. Take a look at https://github.com/mabuchilab/Instrumental/blob/master/instrumental/drivers/cameras/tsi.py to see what's going on.

The driver uses `cffi` to wrap a `tsi_sdk.dll`, which appears to be written in C++ and doesn't expose a C-style ABI. So, it's kinda experimental, but works for me with an 8050M-USB camera (using version 3.3.0.0 of `tsi_sdk.dll`). I've never used a CS2100M-USB before.

danielmtzz, 2018-07-13:
No worries. Thanks for your help.

~Daniel

birdyberth, 2018-09-27:
Hi Daniel, have you managed to get it work? I'm using a CS2100M-USB camera and I'm getting the same issue than you, tsi.list_instruments() returns an empty array.

danielmtzz, 2018-09-27:
Yeah I did, but not with Instrumental. I am using the C API that comes with
the software and ImageMagick to get TIF image files (or basically any
format you want). It’s working great and it’s quite fast. I can send you
the code if you like.

-Daniel

natezb, 2018-09-27:
@danielmtzz Are you referring to the C++ API, or has Thorlabs recently released a C API for its scientific cameras? I'd also be interested in seeing the code.

natezb, 2018-09-27:
I checked on my ThorCam install (v3.0.0), and that SDK directory existed, but only had a single `.cpp` file nested within it.

Installing v3.1.1 just now, there does appear to be a new C API! This is good news as it should let us replace the hacky C++ wrapper with a C-based one using NiceLib. In the 3.1.1 version, the ZIP archive is of `Scientific Camera Interfaces`, not `SDK`.

It also appears (noted in the `C and C++ README` file) that this new SDK supports these new CS-series cameras, which the legacy API did not (which is why they never showed up for either of you).

I'll open an issue aimed at adding support for this new API.

danielmtzz, 2018-09-28:
Here's the code guys @natezb  @birdyberth . Hope it helps. Let me know if there are problems or ways to make it better. It's been a long time since I've written code in plain C.

https://github.com/danielmtzz/ArmCamera_CS2100M-USB

@natezb It would be great to incorporate ImageMagick and the C API with python ;)

birdyberth, 2018-09-29:
Wow thanks a lot Daniel, I'll give it a try!

berto10, 2018-10-18:
Hello I'm trying to run the following camera.  https://www.thorlabs.com/thorproduct.cfm?partnumber=CS505CU 

When I try to view what instruments are connected it sees a camera that uses the UC480 but not the TSI libraries. I added the path as it was mentioned on a previous comment on this thread. However, its still not able to be seen. 

I tried the following and got the message as seen below. 

from instrumental.drivers.cameras import tsi 

[image]

Any suggestion as to what is going on? Thanks!

-Berto

natezb, 2018-10-18:
That error may indicate a bug that's leading to the header file not being properly installed.

However, as I indicated in the other thread, Instrumental does not yet support your camera, which is a Thorlabs "Compact Scientific" camera. The TSI driver only supports Thorlabs "Scientific" cameras.

#64 has more details.

closed 2019-02-12 by natezb