mabuchilab/Instrumental #58: nicelib does not get installed by instrumental's setup.py

jondoesntgit, 2018-07-24:
Just did a clean install on a lab computer, and for importing `instrumental.drivers.daq.ni` I got an  `ImportError: No module named 'nicelib'`

Easily resolved by `pip install nicelib`, but perhaps it would be good to include the dependency in setup.py

jondoesntgit, 2019-03-21:
By the way, here's a stack trace from `pip install instrumental[daq.ni]`

```
pip install instrumental[daq.ni]
Collecting instrumental[daq.ni]
  Downloading https://files.pythonhosted.org/packages/30/03/5872f41363794774d9f71359b12258e709a825125ce3f261adafe07d956d/instrumental-0.5.3.tar.gz (148kB)
    100% |████████████████████████████████| 153kB 22.3MB/s
  instrumental 0.5.3 does not provide the extra 'daq.ni'
Collecting astkit>=0.5.4 (from instrumental[daq.ni])
  Downloading https://files.pythonhosted.org/packages/cc/11/6f67efd895b15345a0a2bbede645b66b40d3242f217a41d0e85205e33a6f/astkit-0.5.4.tar.gz
Collecting mako (from instrumental[daq.ni])
  Downloading https://files.pythonhosted.org/packages/eb/69/6137c60cae2ab8c911bff510bb6d1d23a0189f75d114bb277606c6486b5f/Mako-1.0.8.tar.gz (468kB)
    100% |████████████████████████████████| 471kB 17.8MB/s
Collecting MarkupSafe>=0.9.2 (from mako->instrumental[daq.ni])
  Downloading https://files.pythonhosted.org/packages/98/7b/ff284bd8c80654e471b769062a9b43cc5d03e7a615048d96f4619df8d420/MarkupSafe-1.1.1-cp37-cp37m-manylinux1_x86_64.whl
Building wheels for collected packages: instrumental, astkit, mako
  Building wheel for instrumental (setup.py) ... done
  Stored in directory: /home/daq/.cache/pip/wheels/79/09/f9/ec7d7a0a1a2c77e1b7ad90dd8034d815e1b87caa2b1be85b8c
  Building wheel for astkit (setup.py) ... done
  Stored in directory: /home/daq/.cache/pip/wheels/3e/e5/b5/d384dbdaee3d2e45c25034faedceedf873c2b4088f4a4beadf
  Building wheel for mako (setup.py) ... done
  Stored in directory: /home/daq/.cache/pip/wheels/b9/42/3c/79032e33bb2dfae60a10a6133f828a4a99ae46d33a2d323944
Successfully built instrumental astkit mako
Installing collected packages: astkit, MarkupSafe, mako, instrumental
Successfully installed MarkupSafe-1.1.1 astkit-0.5.4 instrumental-0.5.3 mako-1.0.8
```

natezb, 2019-03-21:
It looks like you accidentally installed `instrumental` rather than `instrumental-lib`. The name clash is unfortunate and sometimes annoying

jondoesntgit, 2019-03-21:
`pip install Instrumental-lib[daq.ni]` works.