mabuchilab/Instrumental #59: Cannot interrogate NI9215 using ni.py out of the box

jondoesntgit, 2018-07-24:
When I try to load the NI9215 DAQ, I get the following stack trace:

```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-4-ef151580f6db> in <module>()
----> 1 daq = instrument(paramsets[0])

c:\users\daq\dropbox\digonnet\instrumental\instrumental\drivers\__init__.py in instrument(inst, **kwargs)
   1265         inst = find_visa_instrument_by_module(params)
   1266     else:
-> 1267         inst = find_nonvisa_instrument(params)
   1268 
   1269     if inst is None:

c:\users\daq\dropbox\digonnet\instrumental\instrumental\drivers\__init__.py in find_nonvisa_instrument(params)
   1165         for classname in classnames:
   1166             try:
-> 1167                 return create_instrument(driver_module, classname, normalized_params)
   1168             except (InstrumentTypeError, InstrumentNotFoundError):
   1169                 log.info("Failed to create instrument using '%s'", classname)

c:\users\daq\dropbox\digonnet\instrumental\instrumental\drivers\__init__.py in create_instrument(driver_module, classname, paramset, visa_inst)
   1085         return cls._create(paramset, _rsrc=visa_inst)
   1086     else:
-> 1087         return cls._create(paramset)
   1088 
   1089 

c:\users\daq\dropbox\digonnet\instrumental\instrumental\drivers\__init__.py in _create(cls, paramset, **other_attrs)
    511         obj._before_init()
    512         obj._fill_out_paramset()
--> 513         obj._initialize(**paramset.get('settings', {}))
    514         obj._after_init()
    515         return obj

c:\users\daq\dropbox\digonnet\instrumental\instrumental\drivers\daq\ni.py in _initialize(self)
   1437         self._load_analog_channels()
   1438         self._load_internal_channels()
-> 1439         self._load_digital_ports()
   1440 
   1441     def _create_mini_task(self, io_type):

c:\users\daq\dropbox\digonnet\instrumental\instrumental\drivers\daq\ni.py in _load_digital_ports(self)
   1459         for line_path in self._dev.GetDevDILines().decode().split(','):
-> 1460             port_name, line_name = line_path.rsplit('/', 2)[-2:]
   1461             if port_name not in ports:
   1462                 ports[port_name] = []

ValueError: not enough values to unpack (expected 2, got 1)
```

jondoesntgit, 2018-07-24:
The issue appears to be that `self._dev.GetDevDILines().decode()` returns a null string (`b''`)

jondoesntgit, 2018-07-24:
Fixed it. Just have to run 

```python
if self._dev.GetDevDILines().decode():
```

before the for loop.

natezb, 2018-07-24:
This was actually recently reported in a pull request #53, and the problem was fixed in a76ef03e411bedd74c1db0029f421bf211b10c2c. Check it out and see if that fixes the issue for you.

jondoesntgit, 2018-07-24:
Gotcha!

I didn't fetch before I installed, and was running off of a month-old clone that I had done way back...

closed 2018-07-24 by jondoesntgit