mabuchilab/Instrumental #61: Loading DAQmx with no AnalogIn channels

Skoricius, 2018-08-14:
I have a DAQmx board which doesn't have any output channels and the NIDAQ class was giving me an error in the _load_internal_channels function from ni.py as it would try to run a for loop over None object:
`1454, in _load_internal_channels
    for ch_name in ch_names:
TypeError: 'NoneType' object is not iterable`

I found a simple fix to be adding a None check before the loop:
```python
def _load_internal_channels(self):
        ch_names = _internal_channels.get(self.product_category)
        if ch_names is not None:
            for ch_name in ch_names:
                setattr(self, ch_name, AnalogIn(self, ch_name))
```

natezb, 2018-08-14:
Thanks for the bug report! Check out 242809061de20e11fa3e36355df6c86e039b96fb, which I just pushed. It should fix this in a slightly more idiomatic way.

What is your DAQ's product category? It's possible that it has internal channels, and we should add them if they're missing.

Skoricius, 2018-09-06:
Thank you! The product is PCIe-6738 (http://www.ni.com/en-gb/support/model.pcie-6738.html)

natezb, 2018-09-07:
I can't find any information suggesting that it has any internal channels. Feel free to re-open this issue if it actually does have them, and you'd like them added.

closed 2018-09-07 by natezb

mankapandey, 2020-12-03:
I'm working with a remote Thorlabs UC480 camera and having a similar error when I go to  - no issue with internal channels, but getting `TypeError: 'NoneType' object is not iterable`. When I define a camera c using `c = instrument(paramsets[0])`, then go to take an image with it `c.grab_image()`, I get the attached traceback. The `c.grab_image()` command has worked for me before, but has been throwing this error recently and I'm not sure how to debug it.

[image]

I'd appreciate any advice in understanding/solving this, thank you in advance!

natezb, 2020-12-15:
Sorry for the delay, and thanks for reporting this. The ultimate error message is the same, but it is a different bug, so I've opened a new issue for this: #125.