mabuchilab/Instrumental #63: Changing Color Gain/White Balance

berto10, 2018-09-22:
Hello, I have a Thorlabs camera and I am trying to change the gain on the individual color. I've was able to successfully change the gain with the following command, but can't figure out how to do it for an individual color. Any help would be appreciated! 

A=cam.grab_image(exposure_time = "32ms",gain = 25)

Thanks! 
-Berto

natezb, 2018-09-27:
Sorry for the delayed response. Setting the gain of individual color channels is not yet implemented in the high-level interface. You should be able to do it using the mid-level interface though:

```
cam._dev.SetHardwareGain(master, red, green, blue)
```

You can can check the "DCx Camera Functional Description and SDK Manual" to confirm, but it appears that these gain values range from 0 to 100.

Auto-whitebalance can be enabled in two slightly-different modes:
```
cam.auto_whitebalance = True
# or
cam.auto_sensor_whitebalance = True
```

If you're interested, feel free to submit a patch adding color-gain to the high-level interface using `Facet`s. Check out how the `master_gain` facet is implemented for clues on how to do this.

berto10, 2018-10-03:
That worked thanks! 

cam._dev.SetHardwareGain(master, red, green, blue)

I just substituted "master", "red", "green", "blue". values from 0-100! 

Thanks,

closed 2018-10-03 by berto10