mabuchilab/Instrumental #66: DAQ Continuous sampling

jondoesntgit, 2018-10-22:
Does Instrumental support continuous sampling?

Couldn't find anything in the docs, and couldn't figure out how to do it from a 30-minute inspection of the codebase. I've done it before using PyDAQmx. I'd be willing to do a pull request if it isn't already supported.

Jonathan

natezb, 2018-10-23:
Yeah, it should, though I haven't used it in several years at least, so I can't attest that it works flawlessly.

When you create a `Task`, you can specify `mode='continuous'` in `set_timing()`. e.g.

```
daq = instrument(...)
task = daq.Task(daq.ai0)
task.set_timing(fsamp='100 Hz', n_samples=1000, mode='continuous')

task.start()
while not done:
    data = task.read()
    ...
task.stop()
```

Let me know if this doesn't work and you can't figure out how to correct it. If you get it working, adding an example to the docs could be valuable.

In general, I like the ergonomics of this module vs PyDAQmx, especially for simple tasks, but there are certainly things that could be cleaned up, added, and made more consistent.

jondoesntgit, 2018-10-23:
I think I missed the “while not done: data=daq.read()” but... let me try that out...

jondoesntgit, 2018-10-23:
It works!

The data that is returned is a dictionary where the keys are paths to channels. Is there a way to specify a name of the channel instead? I think the relevant lines of code are here:

https://github.com/mabuchilab/Instrumental/blob/853af3a6c0df9f6adced646d905f2b354e8531ba/instrumental/drivers/daq/ni.py#L677-L682

jondoesntgit, 2018-10-23:
Workaround:

```python
channels = {
    't': 't',
    'Dev1/ai0': 'lockin_x',
    'Dev1/ai1': 'lockin_y',
    'Dev1/ai2': 'power'
}
data = { channels[k]: v.m for k, v in data.items()}
```

natezb, 2018-10-23:
I believe you can pass in the channel object itself, so you could do something like this:

```
ai_lockin_x = daq.ai0
ai_lockin_y = daq.ai1
ai_power = daq.ai2

power_data = data[ai_power]
```

I could imagine other things we could implement, like creating named copies of a channel, e.g:

```
task = daq.Task(daq.ai0.named('lockin_x'),
                daq.ai1.named('lockin_y'),
                daq.ai2.named('power'))
```

and then named channels could be referenced by name. I'm not sure if it's worth it to add this, but let me know what you think, or if there are other solutions you fancy.

natezb, 2018-12-17:
I'm closing this now, please re-open it if needed.

closed 2018-12-17 by natezb