mabuchilab/Instrumental #67: Problem in thorlabs_css.py

JulesScholler, 2018-12-14:
in function _cdata_to_numpy there is a typo conducting to raising an error, this function should be:

    `def_cdata_to_numpy(self, cdata, data_type=float, size=None):
        if size is None:
            size = self._NiceCCSLib.TLCCS_NUM_PIXELS*BYTES_PER_DOUBLE
        buf = ffi.buffer(ffi.addressof(cdata), size)[:]
        return np.frombuffer(buf, data_type)`

natezb, 2018-12-14:
Can you post the error message here? I can check tomorrow to see if I can reproduce it. In the meantime, yes it would be good to add a Pull Request from your fork.

JulesScholler, 2018-12-14:
I will not have my computer until monday. If I recall correctly it was something related to « buffer » variable not being declared. The right code was « ffi.buffer ». 

The other errors were typos in the class declaration, leading to errors when applying methods in thorlabs_css.py.

natezb, 2018-12-17:
Would you mind posting your error traceback? I haven't been able to reproduce any buffer-related issue on Python 2 or 3. I do see there are some typos related to the external trigger functions, but that's unrelated.

JulesScholler, 2018-12-18:
I see the problem, I installed the module through PIP. If I clone the repo then there is no problem with the function:
### From GitHub repo:
`    def _cdata_to_numpy(self, cdata, data_type=float, size=None):
        if size is None:
            size = self._NiceCCSLib.TLCCS_NUM_PIXELS*BYTES_PER_DOUBLE
        buf = memoryview(ffi.buffer(ffi.addressof(cdata), size)[:])
        return np.frombuffer(buf, data_type)`

Using PIP I had 0.5 instrument-lib with:
### From PIP:
`    def _cdata_to_numpy(self, cdata, data_type=float, size=None):
        if size is None:
            size = self._NiceCCSLib.TLCCS_NUM_PIXELS*BYTES_PER_DOUBLE
        buf = buffer(ffi.buffer(ffi.addressof(cdata), size)[:])
        return np.frombuffer(buf, data_type)`

Which is not working on python 3.6.

natezb, 2018-12-21:
Yeah, there was a fix at some point to handle incompatibilities between py2 and py3 buffers/memoryviews. I'll close this, but feel free to reopen it if you have any more buffer-related issues with this driver.

closed 2018-12-21 by natezb