mabuchilab/Instrumental #69: Unable to set gamma on Thorlabs DCx camera

fauxzor, 2018-12-20:
Hello,

I've recently gotten a Thorlabs DCx (UC480) camera working with Instrumental. I can take data, adjust the ROI, change the gain, all that good stuff. However when I try to set the camera gamma I run into a problem.

```python
from instrumental import instrument, list_instruments
avail = list_instruments()
# Output: [<ParamSet[UC480_Camera] model='C1284R13C' id=1 serial='4103363284'>]
mycam = instrument(avail[0])
# Output: <instrumental.drivers.cameras.uc480.UC480_Camera at 0xa5c99b0>
```

So far, so good, but running `mycam.gamma` gives me the following error:

```python
Traceback (most recent call last):

  File "<ipython-input-5-ae67e1c3ad4e>", line 1, in <module>
    mycam.gamma

  File "C:\Anaconda2\lib\site-packages\instrumental\drivers\__init__.py", line 245, in __get__
    return self.get_value(obj)

  File "C:\Anaconda2\lib\site-packages\instrumental\drivers\__init__.py", line 255, in get_value
    instance.cached_val = self.conv_get(self.fget(obj))

  File "C:\Anaconda2\lib\site-packages\instrumental\drivers\cameras\uc480.py", line 989, in gamma
    return self._dev.Gamma(lib.GAMMA_CMD_GET) / 100.

AttributeError: 'Camera' object has no attribute 'Gamma'
```

Is adjusting the gamma of the DCx cameras supported by Instrumental? I couldn't find any previous issues relating to people being unable to do this so I suspect the problem is on my end. 

For reasons, I am running 32 bit Python 2.7.15 on 64 bit Windows 10. I got Python to find the 32 bit `uc480.dll` required by stealing it from `Thorlabs\Scientific Imaging\DCx Camera Support\OtherDrivers\LabVIEW\For_32-bit_LabVIEW\` and have had success with it so far except with the gamma parameter. Could it be that this is an old version of the library missing the functionality to control the gamma? 

I would be glad to provide any additional information that would be helpful. Thank you for developing this package, it's perfect for what I need to do!

natezb, 2018-12-21:
It looks like a typo on our end: that `Gamma` should instead be `_Gamma`. I'll push a change soon that fixes it. You'll then have to use the dev version from Github (if you aren't already using it), or you can edit the local files to add the fix yourself.

natezb, 2018-12-21:
I just pushed a change that should fix this. Let me know whether or not it fixes your issue, and we can reopen this if needed!

And thanks for reporting!

fauxzor, 2018-12-21:
I installed the latest version from Github and I'm still getting a similar error:

```python
Traceback (most recent call last):

  File "<ipython-input-8-657257b06618>", line 1, in <module>
    mycam.gamma

  File "C:\Anaconda2\lib\site-packages\instrumental_lib-0.6.dev0-py2.7.egg\instrumental\drivers\__init__.py", line 248, in __get__
    return self.get_value(obj)

  File "C:\Anaconda2\lib\site-packages\instrumental_lib-0.6.dev0-py2.7.egg\instrumental\drivers\__init__.py", line 258, in get_value
    instance.cached_val = self.conv_get(self.fget(obj))

  File "C:\Anaconda2\lib\site-packages\instrumental_lib-0.6.dev0-py2.7.egg\instrumental\drivers\cameras\uc480.py", line 989, in gamma
    return self._dev._Gamma(lib.GAMMA_CMD_GET) / 100.

AttributeError: 'Camera' object has no attribute '_Gamma'
```

I don't know if this is useful, but here's what I get from entering `cam` versus `cam._dev`:

```python
mycam._dev
Out[14]: <instrumental.drivers.cameras.uc480.Camera at 0xa3c3bd0>
mycam
Out[15]: <instrumental.drivers.cameras.uc480.UC480_Camera at 0xa3c3df0>
```

Maybe it's asking the right question of the wrong device, or the wrong question of the right device? 

Thank you for such a quick reply!

natezb, 2018-12-22:
Yeah, unfortunately I'm away from the lab right now and couldn't test my "fix" before pushing it. Did you remove the underscore from all locations, including in the definition on line 234?

fauxzor, 2019-01-02:
Sorry for the delay. I was also out of the lab for the holidays. Happy New Year!

I did remove the underscore from line 234 and from the two functions I listed above. That seemed to work on the first machine I tried it with, hence my "I figured it out!" post, but after trying it on another system I still have the same issue as previously:
 ```python 
AttributeError: 'Camera' object has no attribute '_Gamma'
```
even if I rename `_Gamma` to `Gamma` or `Gammadev` or any other name I can think of (keeping it consistent with the definitions in the `gamma` facet. I can start typing `mycam._dev.` and the editor shows me all of its methods, including 'Exposure' and 'Blacklevel' which both work if I type `mycam._dev.Exposure` or whatever. It's almost as if it doesn't 'see' the `Gamma` one even if I change the name.

I did save an instrument when I previously did this fix, so maybe that has something to do with it? I have been trying to use the more general `list_instruments()` approach to find/open the camera with this second machine, which works except for the fact that the gamma issue is back.

commit 9cbc332d (2018-12-21): Fix typo in cameras.uc480 for using gamma

Fixes #69