mabuchilab/Instrumental #70: changing the pixel clock?

spacecrusador, 2019-01-09:
Hello!
I was wondering if there was a way already implented to change the pixel clock (I have a ThorLabs camera, and are therefore using the uc480). Sounds a bit silly as a question since it's a fundamental parameter. Maybe I've missed the way of doing it!

Thank you,
Best

spacecrusador, 2019-01-09:
Rephrased otherwise, when I select a smaller AOI, I can increase the fps but the pixel clock is still the same (24MHz). (I have simply checked the pixel clock by opening the uc480 ThorLabs viewer and found which clock was giving me the minimum and maximum fps).

natezb, 2019-01-09:
There was not pixelclock support, but I just pushed some changes to add it. There's currently no explicit range checking going on, but the underlying function will raise a parameter error if the input is out of range. If needed, it would be easy to add a function/property to get that. For now you could just use:
```
cam = #...open camera however you do
from instrumental.drivers.cameras.uc480 import lib
rng = tuple(cam._dev.PixelClock(lib.PIXELCLOCK_CMD_GET_RANGE))
```
This tuple is (min, max, increment) for the pixelclock in MHz.

Let me what you think, and whether it works for you or not.

spacecrusador, 2019-01-10:
Thank you very much for this! It seems to work fine. Very useful :)

Slighty off topic, to "save a video recording" would you recommend proceeding simply this way:

cam=#...
num_frames=#... number of frames desired
video=np.zeros((num_frames,...))
cam.start_live_video()
kk=0
while kk < num_frames:
    frame_ready = cam.wait_for_frame()
    if frame_ready:
        video[kk] = cam.latest_frame()
        kk=kk+1
cam.stop_live_video()