mabuchilab/Instrumental #74: How to cite you?

aquilesC, 2019-02-03:
Is it possible to cite this package in a scientific publication?

jondoesntgit, 2019-02-03:
I’ve seen some repos publish themselves in the Journal of Open-Source Software (JOSS). Is this something you’ve considered, Nathan?

aquilesC, 2019-02-03:
You may also consider something like [Zenodo](https://zenodo.org/) in case you are interested in collecting citations.

If not, I can put a link to the repo

natezb, 2019-02-03:
I've been meaning to do something like this for awhile, but hadn't gotten around to it. I've just added Instrumental to [Zenodo](https://zenodo.org/record/2556399) (DOI: 10.5281/zenodo.2556399).

JOSS sounds appealing, so it is definitely something I will consider in the near future.

aquilesC, 2019-02-04:
That's great! Thanks for uploading to Zenodo.

I wasn't aware of JOSS and it looks very promising!

closed 2019-02-06 by natezb