mabuchilab/Instrumental #79: Can't import uc480

Slvdr05, 2019-02-12:
Hi! I have trouble importing uc480, I'm running windows 10, python 3 64 bits and have uc480_64.dll both added to my path and copied at the system 32 folder the error i get is the following

`Traceback (most recent call last):
  File "C:\Users\Salvador Alejandro\AppData\Local\Programs\Python\Python37\lib\site-packages\nicelib\__init__.py", line 62, in load_lib
    lib_module = import_module(lib_name, pkg)
  File "C:\Users\Salvador Alejandro\AppData\Local\Programs\Python\Python37\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'instrumental.drivers.cameras._uc480lib'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Salvador Alejandro\AppData\Local\Programs\Python\Python37\lib\site-packages\nicelib\process.py", line 2008, in modify_pattern
    t = next(it)
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Salvador Alejandro\AppData\Local\Programs\Python\Python37\lib\site-packages\nicelib\process.py", line 2014, in modify_pattern
    raise StopIteration
StopIteration

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\Salvador Alejandro\AppData\Local\Programs\Python\Python37\lib\site-packages\instrumental\drivers\cameras\uc480.py", line 32, in <module>
    info = load_lib('uc480', __package__)
  File "C:\Users\Salvador Alejandro\AppData\Local\Programs\Python\Python37\lib\site-packages\nicelib\__init__.py", line 67, in load_lib
    build_module.build(**kwargs)
  File "C:\Users\Salvador Alejandro\AppData\Local\Programs\Python\Python37\lib\site-packages\instrumental\drivers\cameras\_build_uc480.py", line 35, in build
    ignore_system_headers=True)
  File "C:\Users\Salvador Alejandro\AppData\Local\Programs\Python\Python37\lib\site-packages\nicelib\build.py", line 131, in build_lib
    save_dump_file=save_dump_file)
  File "C:\Users\Salvador Alejandro\AppData\Local\Programs\Python\Python37\lib\site-packages\nicelib\process.py", line 1731, in process_headers
    save_dump_file=save_dump_file)
  File "C:\Users\Salvador Alejandro\AppData\Local\Programs\Python\Python37\lib\site-packages\nicelib\process.py", line 1777, in process_source
    header_src, macro_src, tree, argnames = gen.generate()
  File "C:\Users\Salvador Alejandro\AppData\Local\Programs\Python\Python37\lib\site-packages\nicelib\process.py", line 1289, in generate
    tokens = hook(tokens)
  File "C:\Users\Salvador Alejandro\AppData\Local\Programs\Python\Python37\lib\site-packages\nicelib\process.py", line 2034, in add_line_directive_hook
    for t in tokens:
  File "C:\Users\Salvador Alejandro\AppData\Local\Programs\Python\Python37\lib\site-packages\nicelib\process.py", line 2080, in cdecl_hook
    for token in tokens:
  File "C:\Users\Salvador Alejandro\AppData\Local\Programs\Python\Python37\lib\site-packages\nicelib\process.py", line 2273, in stdcall_hook
    token = ph.pop()
  File "C:\Users\Salvador Alejandro\AppData\Local\Programs\Python\Python37\lib\site-packages\nicelib\process.py", line 2127, in pop
    self.peek_deque.append(next(self.tok_it))
RuntimeError: generator raised StopIteration`

I hope you can help me, I'll keep experimenting other options.

natezb, 2019-02-12:
You should try updating to the git version of NiceLib. This appears to be the bug fixed in [mabuchilab/NiceLib@5dc3f75](https://github.com/mabuchilab/NiceLib/commit/5dc3f7594a7fa51312e8183e75e70491c68348de).

Slvdr05, 2019-02-12:
I did it, it worked, thanks a lot, i'll start working with the lib and keep cheking the forum.

closed 2019-02-12 by natezb