mabuchilab/Instrumental #82: EDT pdv open failed error

tang37, 2019-02-18:
Hi,

I am using instrumental to control a Thorlabs DC1545M camera. Though it can run well, but it sometimes gives me ''EDT pdv open failed. Check board installation and unit number, and try restarting the computer''. What does it mean? Is there a way to get rid of that?

I would appreciate it for any help provided,
Yu

natezb, 2019-02-19:
I've seen this before, but I don't know what it means. It would be interesting to hear what the Thorlabs folks have to say about it.

bajabaq, 2019-06-14:
So if you use the latest version of the pdvlib.dll, this message goes away.

I got the 5.5.7.2 version from the EDT website:   https://edt.com/file-category/pdv/   
Downloaded the EDTpdv_5.5.7.2 SDK and installed it.

Then I copied the pdvlib.dll to the ../Thorlabs/Scientific Imaging/ThorCam directory (I renamed the original one to pdvlib-old.dll)

Then you can delete the EDT directory. 

I am using the ThorCam 3.3.0 version of their software, on Windows 10-64, using latest version of instrumental-lib and 64-bit python 3.6.

Hope this helps

natezb, 2019-06-18:
Thanks for the info!

closed 2019-06-18 by natezb

edt-alex, 2024-08-13:
5 years late to the party, I just stumbled across this. One possible reason for this error could be that EDT's kernel driver was not installed or did not find our PCIe frame grabber device. So when software calls our library (userspace API), the `pdv_open()` call can fail with this sort of error message because it cannot talk to our driver attached to the device.

On Windows, this is shown in Device manager. There should be a section for "PCI Devices" and under it you will find an entry for each EDT PCIe device you have installed in the system. 

Running the SDK installer also installs the kernel driver, but it's possible there was a failure in that process somewhere. 

Cheers!