mabuchilab/Instrumental #88: Working with serial instruments in Linux

jondoesntgit, 2019-03-28:
It seems like list_instruments doesn't know how to talk to serial connections on Linux, where you find instruments not through COM3 or COM8, but through `/dev/ttyUSB0`

```
>>> from instrumental import list_instruments
>>> list_instruments()
Traceback (most recent call last):
  File "/home/daq/anaconda3/lib/python3.7/site-packages/serial/serialposix.py", line 265, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '1'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/daq/src/Instrumental/instrumental/drivers/__init__.py", line 844, in list_instruments
    inst_list.extend(list_visa_instruments())
  File "/home/daq/src/Instrumental/instrumental/drivers/__init__.py", line 788, in list_visa_instruments
    return list(gen_visa_instruments())
  File "/home/daq/src/Instrumental/instrumental/drivers/__init__.py", line 741, in gen_visa_instruments
    visa_inst = open_visa_inst(addr, raise_errors=False)
  File "/home/daq/src/Instrumental/instrumental/drivers/__init__.py", line 712, in open_visa_inst
    visa_inst = rm.open_resource(visa_address, open_timeout=50, timeout=200)
  File "/home/daq/anaconda3/lib/python3.7/site-packages/pyvisa/highlevel.py", line 1724, in open_resource
    res.open(access_mode, open_timeout)
  File "/home/daq/anaconda3/lib/python3.7/site-packages/pyvisa/resources/resource.py", line 209, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "/home/daq/anaconda3/lib/python3.7/site-packages/pyvisa/highlevel.py", line 1681, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/home/daq/anaconda3/lib/python3.7/site-packages/pyvisa-py/highlevel.py", line 194, in open
    sess = cls(session, resource_name, parsed, open_timeout)
  File "/home/daq/anaconda3/lib/python3.7/site-packages/pyvisa-py/sessions.py", line 213, in __init__
    self.after_parsing()
  File "/home/daq/anaconda3/lib/python3.7/site-packages/pyvisa-py/serial.py", line 65, in after_parsing
    self.interface = cls(port=self.parsed.board, timeout=self.timeout, write_timeout=self.timeout)
  File "/home/daq/anaconda3/lib/python3.7/site-packages/serial/serialutil.py", line 240, in __init__
    self.open()
  File "/home/daq/anaconda3/lib/python3.7/site-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port 1: [Errno 2] No such file or directory: '1'
```

So, using plain visa, this works:

```
>>> inst = rm.open_resource('ASRL/dev/ttyUSB0::INSTR')
>>> inst.write_termination = '\n'
>>> inst.read_termination = '\n'
>>> self.query('VSET1?')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'self' is not defined
>>> inst.query('VSET1?')
'9.200V\r'
```

But this does not

```
>>> from instrumental.drivers.powersupplies.gw_instek import GPD_3303S
>>> GPD_3303S(visa_address='ASRL/dev/ttyUSB0::INSTR')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/daq/src/Instrumental/instrumental/drivers/__init__.py", line 524, in __new__
    return instrument(inst, **kwds)
  File "/home/daq/src/Instrumental/instrumental/drivers/__init__.py", line 1271, in instrument
    inst = find_visa_instrument(params)
  File "/home/daq/src/Instrumental/instrumental/drivers/__init__.py", line 1057, in find_visa_instrument
    visa_inst = rm.open_resource(visa_address, open_timeout=50, timeout=200)
  File "/home/daq/anaconda3/lib/python3.7/site-packages/pyvisa/highlevel.py", line 1703, in open_resource
    info = self.resource_info(resource_name, extended=True)
  File "/home/daq/anaconda3/lib/python3.7/site-packages/pyvisa/highlevel.py", line 1663, in resource_info
    ret, err = self.visalib.parse_resource_extended(self.session, resource_name)
  File "/home/daq/anaconda3/lib/python3.7/site-packages/pyvisa/ctwrapper/functions.py", line 1383, in parse_resource_extended
    alias_if_exists)
  File "/home/daq/anaconda3/lib/python3.7/site-packages/pyvisa/ctwrapper/highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.
```

natezb, 2019-03-31:
Some log output might be helpful. It's not clear to me offhand why this would be happening, I don't think Instrumental does any special processing of visa addresses.

jondoesntgit, 2019-04-01:
Unplugged the device over the weekend and plugged it back in. Now it's in `/ASRL/dev/ttyUSB1::INSTR/` and everything works. Go figure.

Thanks for your help!

closed 2019-04-01 by jondoesntgit