mabuchilab/Instrumental #91: lextab and yactab littering working directory

jondoesntgit, 2019-05-16:
I'm having an issue in Linux (CentOS7) where the lextab and yactab get dumped into the working directory.

I saw somewhere where there was a hack to try to prevent this from happening, but I'm not using the camera.

https://github.com/mabuchilab/Instrumental/blob/08c789fd0d109a40bbf8ffa62df2df2bc9dd4c3d/instrumental/drivers/cameras/pco.py#L45-L49

I can try to find the root cause, and see what I can do to fix it. Do you have any guidance before I start?

natezb, 2019-05-16:
You're right, this has been an issue in the past, but I *think* it should be fixed in the current `master` of NiceLib. Check out mabuchilab/NiceLib@eb0b2409ed3f for the most recent reference I could find.

Ultimately the issue comes from NiceLib's use of pycparser. It will generate these tables if it doesn't find them in the install directory (usually site-packages, but could be you dev directory if you're using `pip -e`). So, if you're still seeing this issue with an up-to-date NiceLib, you might investigate why these tables aren't being properly installed, or why they can't be found.

NiceLib has largely copied pycparser's technique of building and installing the tables within `setup.py` during an install.

zakv, 2019-12-11:
I ran into this same issue recently. I suspect that the issue may be that the code in NiceLib's `setup.py` isn't run if NiceLib is installed via pip rather than from source. The relevant part here is that `setup.py` calls `_build_tables.py`, which is located in NiceLib's parser subdirectory. The workaround that worked for me was to manually run `_build_tables.py`.

To do this on Windows, open a shell, cd to the directory containing `_build_tables.py` (for me it was `C:\Users\Username\Software\Anaconda3\Lib\site-packages\nicelib\parser`) and run that script

```cmd
cd C:\Users\Username\Software\Anaconda3\Lib\site-packages\nicelib\parser
python .\_build_tables.py
```

Running the above created `lextab.py` and `yacctab.py` in the directory with `_build_tables.py`. After that, those files stopped being generated in my working directory when using NiceLib.