mabuchilab/Instrumental #92: problems using thorlabs_ccs.py

jrr1984, 2019-05-17:
Hi!

Im using your following script: https://github.com/mabuchilab/Instrumental/blob/master/instrumental/drivers/spectrometers/thorlabs_ccs.py

I installed the Instrumental package and also i copied the whole folder of instrumental and run the script in the same directory where is located in your package.

When i run the code of thorlabs_css.py i get the following error:

---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-3-8d97ff6b7593> in <module>
     14 from nicelib import NiceLib, Sig, NiceObject, load_lib, RetHandler, ret_ignore
     15 
---> 16 from . import Spectrometer
     17 from ..util import check_units, check_enums
     18 from .. import ParamSet

ImportError: cannot import name 'Spectrometer' from '__main__' (unknown location)

Im sorry if this is too simple/foolish. Do you have any example code of how to use your class of the spectrometer of thorlabs (i got the CCS200/M)?

Thanks in advance!

natezb, 2019-05-17:
Instrumental is designed to be used as a library, not a script, so you should avoid running code within the package directory. The error you are seeing is because of this. To ensure you have all the python dependencies for the CCS driver, you can specify it as an "extra":
```
pip install instrumental-lib[spectrometers.thorlabs_ccs]
```
Note that you may need to have other non-python dependencies installed too, like the Thorlabs CCS software.

Once everything is installed, you can use `list_instruments()` to check if your instrument has been located
```
from instrumental import list_instruments
list_instruments()
```

There is additional info in our [documentation](https://instrumental-lib.readthedocs.io/en/stable/instruments.html) on how to open and use instruments. If `list_instruments()` is not locating the instrument, you may want to enable logging (as described on the linked page) to help debug.

I'm not aware of any CCS-specific code examples, but the API documentation for the `CCS` class is located [here](https://instrumental-lib.readthedocs.io/en/stable/spectrometers-thorlabs_ccs.html#instrumental.drivers.spectrometers.thorlabs_ccs.CCS)

jrr1984, 2019-05-22:
Thank you very much for your answer.

I tried the following:

from instrumental import instrument, list_instruments
from instrumental.drivers.spectrometers.thorlabs_ccs import CCS
paramsets = list_instruments()
paramsets

Got this:

[<ParamSet[CCS] usb='USB0::0x1313::0x8089::M00417163::RAW' serial='M00417163' model=<SpecTypes.CCS200: 32905>>]

and when i tried to create the instrument:

ccs200 = instrument(paramsets[0])
 
i got the following error, any clue? thanks in advance.

---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
<ipython-input-11-4028cce4d8f9> in <module>
----> 1 ccs200 = instrument(paramsets[0])

C:\ProgramData\Anaconda3\lib\site-packages\instrumental\drivers\__init__.py in instrument(inst, **kwargs)
   1265         inst = find_visa_instrument_by_module(params)
   1266     else:
-> 1267         inst = find_nonvisa_instrument(params)
   1268 
   1269     if inst is None:

C:\ProgramData\Anaconda3\lib\site-packages\instrumental\drivers\__init__.py in find_nonvisa_instrument(params)
   1165         for classname in classnames:
   1166             try:
-> 1167                 return create_instrument(driver_module, classname, normalized_params)
   1168             except (InstrumentTypeError, InstrumentNotFoundError):
   1169                 log.info("Failed to create instrument using '%s'", classname)

C:\ProgramData\Anaconda3\lib\site-packages\instrumental\drivers\__init__.py in create_instrument(driver_module, classname, paramset, visa_inst)
   1085         return cls._create(paramset, _rsrc=visa_inst)
   1086     else:
-> 1087         return cls._create(paramset)
   1088 
   1089 

C:\ProgramData\Anaconda3\lib\site-packages\instrumental\drivers\__init__.py in _create(cls, paramset, **other_attrs)
    511         obj._before_init()
    512         obj._fill_out_paramset()
--> 513         obj._initialize(**paramset.get('settings', {}))
    514         obj._after_init()
    515         return obj

C:\ProgramData\Anaconda3\lib\site-packages\instrumental\drivers\spectrometers\thorlabs_ccs.py in _initialize(self)
    158         self._NiceCCSLib = NiceCCSLib
    159         self._open(self._address)
--> 160         self._wavelength_array = self.calibrate_wavelength(calibration_type=Calibration.Factory)
    161 
    162     def __del__(self):

<decorator-gen-2> in calibrate_wavelength(self, calibration_type, wavelength_array, pixel_array)

C:\ProgramData\Anaconda3\lib\site-packages\instrumental\drivers\util.py in wrapper(func, *args, **kwds)
    197                 checked[name] = checkers[name](arg) if name in checkers else arg
    198 
--> 199             result = func(**checked)
    200             return result
    201         return decorator.decorate(func, wrapper)

C:\ProgramData\Anaconda3\lib\site-packages\instrumental\drivers\spectrometers\thorlabs_ccs.py in calibrate_wavelength(self, calibration_type, wavelength_array, pixel_array)
    419             self._NiceCCS.setWavelengthData(pixel_array, wavelength_array, num_points)
    420         wavelength_array, _, _ = self._NiceCCS.getWavelengthData(calibration_type.value)
--> 421         self._wavelength_array = self._cdata_to_numpy(wavelength_array)
    422         return self._wavelength_array
    423 

C:\ProgramData\Anaconda3\lib\site-packages\instrumental\drivers\spectrometers\thorlabs_ccs.py in _cdata_to_numpy(self, cdata, data_type, size)
    291         if size is None:
    292             size = self._NiceCCSLib.TLCCS_NUM_PIXELS*BYTES_PER_DOUBLE
--> 293         buf = buffer(ffi.buffer(ffi.addressof(cdata), size)[:])
    294         return np.frombuffer(buf, data_type)
    295 

NameError: name 'buffer' is not defined

natezb, 2019-05-22:
This appears to be a bug fixed in 534f45d6f2f872116b094f1b9841fb7096201f31. Unfortunately this has not been released yet, so you'll have to use the version from github. There is information [here](https://instrumental-lib.readthedocs.io/en/stable/install.html#installing-the-development-version-from-github) on how to do this.

Also beware that when I tested installing the `spectrometers.thorlabs_ccs` "extra", it erroneously installed the `visa` package from PyPI rather than `pyvisa`. So if you get any errors related to importing from visa, uninstall `visa` (`pip uninstall visa`) and install pyvisa (`pip install pyvisa`).

jrr1984, 2019-05-23:
Thank you very much for your help.

I installed the github version and now when i just run:
from instrumental import instrument, list_instrument

i got the following error, any clue? Thanks in advance.

Traceback (most recent call last):

  File "C:\Users\juanr\.conda\envs\aloha\lib\site-packages\IPython\core\interactiveshell.py", line 3296, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)

  File "<ipython-input-1-1863459cd715>", line 1, in <module>
    from instrumental import instrument, list_instrument

  File "<frozen importlib._bootstrap>", line 1032, in _handle_fromlist

  File "C:\Users\juanr\.conda\envs\aloha\lib\site-packages\instrumental_lib-0.6.dev0-py3.7.egg\instrumental\__init__.py", line 50, in __getattr__
    module = __import__(object_origins[name], None, None, [name])

  File "C:\Users\juanr\.conda\envs\aloha\lib\site-packages\instrumental_lib-0.6.dev0-py3.7.egg\instrumental\drivers\__init__.py", line 24, in <module>
    from .. import conf

  File "<frozen importlib._bootstrap>", line 1032, in _handle_fromlist

  File "C:\Users\juanr\.conda\envs\aloha\lib\site-packages\instrumental_lib-0.6.dev0-py3.7.egg\instrumental\__init__.py", line 55, in __getattr__
    __import__('instrumental.' + name)

  File "C:\Users\juanr\.conda\envs\aloha\lib\site-packages\instrumental_lib-0.6.dev0-py3.7.egg\instrumental\conf.py", line 91, in <module>
    load_config_file()

  File "C:\Users\juanr\.conda\envs\aloha\lib\site-packages\instrumental_lib-0.6.dev0-py3.7.egg\instrumental\conf.py", line 69, in load_config_file
    d = literal_eval(value)

  File "C:\Users\juanr\.conda\envs\aloha\lib\ast.py", line 46, in literal_eval
    node_or_string = parse(node_or_string, mode='eval')

  File "C:\Users\juanr\.conda\envs\aloha\lib\ast.py", line 35, in parse
    return compile(source, filename, mode, PyCF_ONLY_AST)

  File "<unknown>", line 1
    {'usb': 'USB0::0x1313::0x8089::M00417163::RAW', 'serial': 'M00417163', 'model': <SpecTypes.CCS200: 32905>, 'module': 'spectrometers.thorlabs_ccs', 'classname': 'CCS'}
                                                                                    ^
SyntaxError: invalid syntax

natezb, 2019-05-23:
It looks like you have an instrument that was saved to your `instrumental.conf` file in an invalid way, leading to a failed load of the config file. Seems like a bug in the CCS driver or `save_instrument` code.

In the meantime, you can probably manually edit the config file in a text editor, and remove the problematic `'model': <SpecTypes.CCS200: 32905>,` entry from its dictionary. The config file is on your system is probably `C:\Users\juanr\AppData\Local\Mabuchilab\Instrumental\instrumental.conf`.

Also note that the function is `list_instruments`, not `list_instrument`.

jrr1984, 2019-05-23:
Thats great, solved the issue! thanks a lot for your time and patience.

When i installed the package cloning the repo and running python setup.py install i got at the end the following error:

No C compiler was found, so cffi modules were not built. If you would like to use cffi-based drivers that require compilation, first install a suitable compiler, then reinstall Instrumental. See the cffi installation documentation for more details on installing an appropriate compiler for your platform.

i think that-s causing that some methods of the class CCS don-t work.. i got a c compiler, installed MINGW, do you know why is giving me that error? Thanks again in advance.

natezb, 2019-05-23:
That warning doesn't apply to your case, as the CCS driver should not require a compiler, only the pixelfly driver does.

Are there specific errors you get when trying to use the CCS class?

jrr1984, 2019-05-23:
Everything is working great, out of the box! Thank you very much for your
help!

natezb, 2019-05-23:
Great to hear! Feel free to file another issue if you come up against any other problems or bugs.

closed 2019-05-23 by natezb