mabuchilab/Instrumental #93: method take_data thorlabs spectrometer

jrr1984, 2019-05-24:
Hi! Sorry to bother again, im getting constantly the following error when running take_data for the thorlabs spectrometer ccs200/m

---------------------------------------------------------------------------
ThorlabsCCSError                          Traceback (most recent call last)
<ipython-input-12-9e27de58db4b> in <module>()
----> 1 intensidad,long_de_onda = ccs.take_data(integration_time=None, num_avg=1, use_background=False)
      2 #plt.plot(long_de_onda,intensidad)

C:\ProgramData\Anaconda3\envs\syst\lib\site-packages\instrumental_lib-0.6.dev0-py3.6.egg\instrumental\drivers\spectrometers\thorlabs_ccs.py in take_data(self, integration_time, num_avg, use_background)
    335             The wavelength (in nm) corresponding to each pixel.
    336         """
--> 337         self.stop_and_clear()
    338         if integration_time is not None:
    339             self.set_integration_time(integration_time)

C:\ProgramData\Anaconda3\envs\syst\lib\site-packages\instrumental_lib-0.6.dev0-py3.6.egg\instrumental\drivers\spectrometers\thorlabs_ccs.py in stop_and_clear(self)
    309         self.stop_scan()
    310         while self.is_data_ready():
--> 311             self.get_scan_data()
    312 
    313     def take_data(self, integration_time=None, num_avg=1, use_background=False):

C:\ProgramData\Anaconda3\envs\syst\lib\site-packages\instrumental_lib-0.6.dev0-py3.6.egg\instrumental\drivers\spectrometers\thorlabs_ccs.py in get_scan_data(self)
    285         data : numpy array of type float with of length NUM_RAW_PIXELS = 3648,
    286         """
--> 287         data = self._NiceCCS.getScanData()
    288         return self._cdata_to_numpy(data)
    289 

C:\ProgramData\Anaconda3\envs\syst\lib\site-packages\nicelib\nicelib.py in __call__(self, *args)
    691 
    692     def __call__(self, *args):
--> 693         return self._libfunc(*(self._niceobj._handles + args), niceobj=self._niceobj)
    694 
    695 

C:\ProgramData\Anaconda3\envs\syst\lib\site-packages\nicelib\nicelib.py in __call__(self, *args, **kwds)
   1151             'funcargs': c_args,
   1152         }
-> 1153         return self.sig.extract_outputs(c_args, retval, ret_handler_args)
   1154 
   1155 

C:\ProgramData\Anaconda3\envs\syst\lib\site-packages\nicelib\nicelib.py in extract_outputs(self, c_args, retval, ret_handler_kwargs)
    212 
    213         if self.ret_handler:
--> 214             retval = self.ret_handler.handle(retval, ret_handler_kwargs)
    215 
    216         if retval is not None:

C:\ProgramData\Anaconda3\envs\syst\lib\site-packages\nicelib\nicelib.py in handle(self, retval, available_kwargs)
    535             raise KeyError("Unknown arg '{}' in arglist of ret-handling function "
    536                            "'{}'".format(e.args[0], self.__name__))
--> 537         return self.__func__(retval, **kwargs)
    538 
    539 

C:\ProgramData\Anaconda3\envs\syst\lib\site-packages\instrumental_lib-0.6.dev0-py3.6.egg\instrumental\drivers\spectrometers\thorlabs_ccs.py in ccs_errcheck(error_code, niceobj)
     54             raise ThorlabsCCSError(NiceCCSLib.error_message(0, error_code)[0])
     55         else:
---> 56             raise ThorlabsCCSError(niceobj.error_message(error_code)[0])
     57 
     58 

ThorlabsCCSError: 84

Do you have any idea what is the problem? Thanks in advance.

natezb, 2019-05-24:
I didn't write this driver and haven't used it, so I can't speak from experience. However, it appears the underlying `getScanData()` function in the TLCCS library is returning an error cod 84. You should check with the TLCCS documentation to see if they provide any information about this error code.

Someone else chime in if you've seen this before.

rogers132, 2019-05-25:
Just checked and the current version works for me with python 3.7 and a CCS175 spectrometer.  From the traceback it looks like you might not be using the most recent version of the driver.  I have also fixed a bug in the error reporting, so there might be a more illuminating error message if you reinstall with the most recent github version and try again.  Can you properly connect to the spectrometer using the Thorlabs OSA software?

The code I used was:
```python
from instrumental import instrument
ccs = instrument('CCS', serial='XXXX')
a  = ccs.take_data()
```

Note that sometimes the first time after the spectrometer was plugged in ``` ccs.take_data()``` would time out, but after that first time it would always work fine.

jrr1984, 2019-05-25:
Thank you for your reply rogers132! Unfortunately the method take_data workings randomsly.. sometimes it works, sometimes not.. It still giving me i think the same error:

---------------------------------------------------------------------------
ThorlabsCCSError                          Traceback (most recent call last)
<ipython-input-49-4bcbc43bac86> in <module>
----> 1 y,x = ccs.take_data(integration_time=None, num_avg=1, use_background=False)

C:\ProgramData\Anaconda3\envs\syst3.7\lib\site-packages\instrumental_lib-0.6.dev0-py3.7.egg\instrumental\drivers\spectrometers\thorlabs_ccs.py in take_data(self, integration_time, num_avg, use_background)
    363                 raise Warning('Raw data is saturated')
    364 
--> 365         self.stop_and_clear()
    366         data = data/num_avg
    367         if use_background:

C:\ProgramData\Anaconda3\envs\syst3.7\lib\site-packages\instrumental_lib-0.6.dev0-py3.7.egg\instrumental\drivers\spectrometers\thorlabs_ccs.py in stop_and_clear(self)
    315         self.stop_scan()
    316         while self.is_data_ready():
--> 317             self.get_scan_data()
    318 
    319     def take_data(self, integration_time=None, num_avg=1, use_background=False):

C:\ProgramData\Anaconda3\envs\syst3.7\lib\site-packages\instrumental_lib-0.6.dev0-py3.7.egg\instrumental\drivers\spectrometers\thorlabs_ccs.py in get_scan_data(self)
    291         data : numpy array of type float with of length NUM_RAW_PIXELS = 3648,
    292         """
--> 293         data = self._NiceCCS.getScanData()
    294         return self._cdata_to_numpy(data)
    295 

C:\ProgramData\Anaconda3\envs\syst3.7\lib\site-packages\nicelib\nicelib.py in __call__(self, *args)
    691 
    692     def __call__(self, *args):
--> 693         return self._libfunc(*(self._niceobj._handles + args), niceobj=self._niceobj)
    694 
    695 

C:\ProgramData\Anaconda3\envs\syst3.7\lib\site-packages\nicelib\nicelib.py in __call__(self, *args, **kwds)
   1151             'funcargs': c_args,
   1152         }
-> 1153         return self.sig.extract_outputs(c_args, retval, ret_handler_args)
   1154 
   1155 

C:\ProgramData\Anaconda3\envs\syst3.7\lib\site-packages\nicelib\nicelib.py in extract_outputs(self, c_args, retval, ret_handler_kwargs)
    212 
    213         if self.ret_handler:
--> 214             retval = self.ret_handler.handle(retval, ret_handler_kwargs)
    215 
    216         if retval is not None:

C:\ProgramData\Anaconda3\envs\syst3.7\lib\site-packages\nicelib\nicelib.py in handle(self, retval, available_kwargs)
    535             raise KeyError("Unknown arg '{}' in arglist of ret-handling function "
    536                            "'{}'".format(e.args[0], self.__name__))
--> 537         return self.__func__(retval, **kwargs)
    538 
    539 

C:\ProgramData\Anaconda3\envs\syst3.7\lib\site-packages\instrumental_lib-0.6.dev0-py3.7.egg\instrumental\drivers\spectrometers\thorlabs_ccs.py in ccs_errcheck(error_code, niceobj)
     60             raise ThorlabsCCSError(NiceCCSLib.error_message(0, error_code))
     61         else:
---> 62             raise ThorlabsCCSError(niceobj.error_message(error_code))
     63 
     64 

ThorlabsCCSError: b'Timeout expired before operation completed.'

the code i used was the same you gave me but im trying to call take_data several times cause i-m willing to do a spatial scan so i have to do several spectral measurements.

Im using the ccs200/m spectrometer, OSA software version 2.9, windows 10 64 bits, python 3.7 conda distribution.

Have you tried to run the method several times and it works just fine? Thank you very much in advance.

rogers132, 2019-05-28:
Hi jrr1984, thanks for the new info.

I have made some changes to the driver to make it more robust against timeout errors, so reinstall with the latest version and give that a try.

To answer your question, yes I have run the method several times using the original driver and it usually works fine on my system.  In some cases I have run it >10000 times without errors.

It looks like this is a timeout error from the spectrometer itself, which means it could just be due to an intermittent connection issue between your spectrometer and computer.  Its hard to say for sure at this point though.  In the past I have seen some issues with timeout errors like yours, but after swapping the usb cable they went away.

Also, does the spectrometer work fine using the thorlabs OSA GUI software?  And what integration time are you using for each scan?

jrr1984, 2019-05-29:
Thank you very much for your reply and for the updates. Im gonna try it tomorrow and also try with another usb cable as well.

The spectrometer works fine with the OSA soft. I'm using 1 milisecond of integration time.

closed 2019-06-17 by rogers132