mabuchilab/Instrumental #95: Linux support for Thorlabs DCx Camera

fauxzor, 2019-06-19:
This is not strictly speaking an "issue" as it is pretty explicit in the [documentation here](https://instrumental-lib.readthedocs.io/en/stable/uc480-cameras.html) that Linux support is not implemented yet, but in my mind I can interpret this as belonging here.

I have had great success controlling a Thorlabs DCx camera using Instrumental on a Windows machine, but now I have a need to make it work on a 64-bit Linux machine. Broadly speaking I've run into two kinds of issues:

- Figuring out the correct header & API to use
- Getting them to work with `nicelib`

I was able to find Linux drivers for the camera from Thorlabs [at this link (under 'Drivers')](https://www.thorlabs.com/software_pages/viewsoftwarepage.cfm?code=ThorCam) which are identical to those provided by IDS for their uEye cameras (which Thorlabs rebranded to make the DCx series as I understand). By following the [instructions here](https://www.thorlabs.com/software/THO/ThorCam/Linux/Readme_uEye_Linux_4.90.06.htm) I was able to open the camera on the Linux machine in the uEye camera controller.

However, it appears that `pywin32` is needed for `uc480.py` to do its thing and this is about where my understanding starts to deteriorate. It doesn't look like `pywin32` does all that much so it should be replaceable with something but I am not sure what the equivalent is for Linux (if there is one).

After pointing `_build_uc480.py` to the "correct" version of the Linux API & header (`libueye_api64.so` and `ueye.h`), `nicelib` also fails parsing any line beginning with `IDSEXP` in `ueye.h`, which appears to replace the `USBCAMEXP` syntax in `uc480.h`. I tried mix-and-matching them by pairing `uc480.h` with `libueye_api64.so` and `nicelib` did compile something, but the resulting `_uc480lib.py` file didn't work.

I would be glad to provide any error messages or additional info if necessary but I believe my problem is mostly conceptual. If it's possible to answer this without actually implementing it yourself, what would need to be done to control a uc480 camera in Linux?

natezb, 2019-06-21:
Thanks for your patience, I've been a bit busy recently. I can take a closer look tomorrow, but here's what I remember off the top of my head. The APIs for Linux and Windows are *mostly* the same, but not exactly. Where they differ is primarily (or maybe entirely?) with the event handling. The Windows version uses the Windows event system (e.g. `WaitForSingleObject()`), and I don't remember what the Linux version uses. Look into the event-related section of the ThorLabs/IDS docs to find more info.

`pywin32` was originally used to interface with this Windows event API, but that was before this driver got ported to use NiceLib. It should be possible to eliminate the `pywin32` requirement for Windows, but it might take a little bit of work.

As for the NiceLib stuff, you might try posting the error traceback and I could give you a suggestion. There's probably something that needs to be stripped out of the header.

bvieira13, 2023-11-20:
Hi, i try to control a Thorlabs DCx camera in Linux too. However, I don't understood how to make this.
Can tell if it was implemented for Linux OS?