mabuchilab/Instrumental #101: thorlabs k10cr1 issues

TraceBivens, 2019-09-30:
I'm having several issues with controlling k10cr1 rotation stages. The least important of these is that I have to use a try/except block to call instrument('saved_instrument') twice for each stage in order to actually load the things, the first call to instrument() raises a 'no known Paramset' error. 

More importantly, I'm getting different behavior from different stages. We have 3 K10CR1 stages, one that we've had for a while and a pair that we just got in. In the case of the older one, stage.home() works as expected after an offset has been defined, but the stage always reports position relative to the zero point when it was initialized; this means that move_to() does *not* account for the offset.

After determining this, I started working on one of the newer stages and discovered that stage.home() doesn't appear to work _at _all_ in this case; the active light comes on, but the stage doesn't move.

natezb, 2019-10-02:
Thanks for the report. What version of Instrumental and NiceLib are you using?

I'd be interested to see the traceback you get when trying to open the instrument, and (maybe more usefully) your `instrumental.conf` file.

I haven't seen the issue with not-homing before. Maybe the new stage is only compatible with a newer version of the Thorlabs Kinesis API?

Also, you can look at the code for the K10CR1 and search for `offset` to see how it's used internally. I haven't thought about this stuff in awhile, but how do you expect the offset should work?

TraceBivens, 2019-10-02:
So I made some progress since I posted this and figured out that in fact the stages *are* behaving the same, but I was doing this in the interpreter to test before running my experiment and I was getting confused keeping track of the offsets and positions.

I don’t have the version information or traceback on hand right now, but I can get those when I get into the lab in the morning. I can say that I installed instrumental and nicelib via pip sometime in the last month or so.

I did find a workaround by using a try/except block that just calls instrument(‘saved_instrument’) twice.

The issue with homing appears to be related to wait_for_home(); if I just tell the stages to go where I want them to go, they work fine, but if I call home() with wait=True, they seem to get locked in an endless loop. No motion, but the active light stays on.

My expectation for offset was based on the documentation saying that move_to() accounts for  the offset, leading me to believe that after setting the offset the stage would report its position as zero when homed to the offset point. In practice, after an offset is set and the stage is homed the stage reports its position as the negative of the offset, and continues to report its position relative to the zero point on startup, so when I want it to go to a position relative to the offset I have to subtract the offset from the position I’m feeding to move_to(). While trying to understand this behavior I pulled up the kinesis GUI and noticed that the offset value in the preferences is greyed out and not editable; not sure whether that’s related or not but figured it might be worth mentioning.

Also, this may just be due to inexperience but the documentation says that move_to() takes its argument in degrees; my naive expectation was that I would simply pass an int as the argument and that it would interpret that as degrees, but in practice doing that makes it move in radians. Using ureg.degree works as expected, or passing a string like move_to(‘10 degree’), but I spent a solid half hour being confused by that before it occurred to me that pint might solve the issue.

TraceBivens, 2019-10-03:
Here's an example including the traceback. 

Using NiceLib version 0.6
Instrumental version 0.5

------------------------------------
Python 3.7.3 | packaged by conda-forge | (default, Jul  1 2019, 22:01:29) [MSC v.1900 64 bit (AMD64)]
Type "copyright", "credits" or "license" for more information.

IPython 7.8.0 -- An enhanced Interactive Python.

from instrumental import instrument

rotator_top = instrument('rotator_top')
Traceback (most recent call last):

  File "<ipython-input-2-1b38e9fe6e86>", line 1, in <module>
    rotator_top = instrument('rotator_top')

  File "C:\ProgramData\Miniconda3\envs\autobot\lib\site-packages\instrumental\drivers\__init__.py", line 1267, in instrument
    inst = find_nonvisa_instrument(params)

  File "C:\ProgramData\Miniconda3\envs\autobot\lib\site-packages\instrumental\drivers\__init__.py", line 1167, in find_nonvisa_instrument
    return create_instrument(driver_module, classname, normalized_params)

  File "C:\ProgramData\Miniconda3\envs\autobot\lib\site-packages\instrumental\drivers\__init__.py", line 1087, in create_instrument
    return cls._create(paramset)

  File "C:\ProgramData\Miniconda3\envs\autobot\lib\site-packages\instrumental\drivers\__init__.py", line 513, in _create
    obj._initialize(**paramset.get('settings', {}))

  File "<decorator-gen-1>", line 2, in _initialize

  File "C:\ProgramData\Miniconda3\envs\autobot\lib\site-packages\instrumental\drivers\util.py", line 293, in wrapper
    result = func(*new_args, **new_kwargs)

  File "C:\ProgramData\Miniconda3\envs\autobot\lib\site-packages\instrumental\drivers\motion\kinesis.py", line 212, in _initialize
    self._wait_for_message(GenericDevice.SettingsInitialized)

  File "C:\ProgramData\Miniconda3\envs\autobot\lib\site-packages\instrumental\drivers\motion\kinesis.py", line 261, in _wait_for_message
    msg_id, msg_data = self._decode_message(self.dev.WaitForMessage())

  File "C:\ProgramData\Miniconda3\envs\autobot\lib\site-packages\instrumental\drivers\motion\kinesis.py", line 254, in _decode_message
    msg_id = MessageIDs[msg_type](msg_id_int)

  File "C:\ProgramData\Miniconda3\envs\autobot\lib\enum.py", line 310, in __call__
    return cls.__new__(cls, value)

  File "C:\ProgramData\Miniconda3\envs\autobot\lib\enum.py", line 564, in __new__
    raise exc

  File "C:\ProgramData\Miniconda3\envs\autobot\lib\enum.py", line 548, in __new__
    result = cls._missing_(value)

  File "C:\ProgramData\Miniconda3\envs\autobot\lib\enum.py", line 577, in _missing_
    raise ValueError("%r is not a valid %s" % (value, cls.__name__))

ValueError: 5 is not a valid GenericDevice
-----------------------------------------------

Here's the config file:

----------------------------
# Instrumental config file
#
# Settings are organized under sections denoted by headings [section name]
# Within each section, settings are written in key = value form.
# Don't use quotation marks around the values.

[servers] #--------------------------------------------------------------------
# This section lists available Instrumental servers and their addresses in
# host:port form. The keys are server aliases.

#myServer = 0.0.0.0:28265

[instruments] #----------------------------------------------------------------
# This section includes instruments and their parameters. The keys are aliases,
# and the values are written as python dicts of instrument parameters. See the
# examples below for more info.

# VISA instruments
#myScope = {'visa_address': 'TCPIP::0.0.0.1::INSTR'}
#myAFG = {'visa_address': 'TCPIP::0.0.0.2::INSTR'}

# NI-DAQ device
#myDAQ = {'nidaq_devname': 'Dev1'}

# Thorlabs/uEye camera
#myCam = {'ueye_cam_id': 2}

# Entry auto-created 2019-09-25 13:26:58
attenuator = {'serial': '55001000', 'module': 'motion.kinesis', 'classname': 'K10CR1'}

# Entry auto-created 2019-09-25 13:33:35
rotator_top = {'serial': '55114554', 'module': 'motion.kinesis', 'classname': 'K10CR1'}

# Entry auto-created 2019-09-25 13:33:50
rotator_bottom = {'serial': '55114654', 'module': 'motion.kinesis', 'classname': 'K10CR1'}

[prefs] #----------------------------------------------------------------------
# This section is for miscellaneous user-specific preferences.

# This is a path to the root directory where data files will be saved
data_directory = ~/Data
-----------------------------------------

TraceBivens, 2019-10-03:
Sorry about the formatting, i'm kinda new to github and still figuring out how to get stuff to look right