mabuchilab/Instrumental #102: Thorlabs Spectrometer issues

eveloen, 2019-10-08:
Hi,
I'm having an issue getting the my thorlabs spectrometer to work with the library.

I am doing the following: 
`import instrumental
from instrumental import instrument, list_instruments

paramsets = list_instruments()
print(paramsets)
my_CCS = instrument(paramsets[0])`

The list_instrument() statement seems to work as it gives me 
[<ParamSet[CCS] model=<SpecTypes.CCS175: 32903> serial=u'M00411244' usb=u'USB0::0x1313::0x8087::M00411244::RAW'>]

Calling instrument() throws the following exception: AttributeError: "'CCS' object has no attribute '_NiceCCS'"
Full traceback
`Traceback (most recent call last):
  File "c:\Users\eweiss\.vscode\extensions\ms-python.python-2019.9.34911\pythonFiles\ptvsd_launcher.py", line 43, in <module>
    main(ptvsdArgs)
  File "c:\Users\eweiss\.vscode\extensions\ms-python.python-2019.9.34911\pythonFiles\lib\python\ptvsd\__main__.py", line 432, in main
    run()
  File "c:\Users\eweiss\.vscode\extensions\ms-python.python-2019.9.34911\pythonFiles\lib\python\ptvsd\__main__.py", line 316, in run_file
    runpy.run_path(target, run_name='__main__')
  File "C:\Users\eweiss\.conda\envs\IPAEM\lib\runpy.py", line 252, in run_path
    return _run_module_code(code, init_globals, run_name, path_name)
  File "C:\Users\eweiss\.conda\envs\IPAEM\lib\runpy.py", line 82, in _run_module_code
    mod_name, mod_fname, mod_loader, pkg_name)
  File "C:\Users\eweiss\.conda\envs\IPAEM\lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "p:\gsg\gsgstud\07_Interdisciplinary_Project\IPA_Meyer_Weiss\10_Scratch\python\thorlabs_instrumental-lib.py", line 27, in <module>
    my_CCS = instrument(paramsets[0])
  File "C:\Users\eweiss\.conda\envs\IPAEM\lib\site-packages\instrumental\drivers\__init__.py", line 1267, in instrument
    inst = find_nonvisa_instrument(params)
  File "C:\Users\eweiss\.conda\envs\IPAEM\lib\site-packages\instrumental\drivers\__init__.py", line 1167, in find_nonvisa_instrument
    return create_instrument(driver_module, classname, normalized_params)
  File "C:\Users\eweiss\.conda\envs\IPAEM\lib\site-packages\instrumental\drivers\__init__.py", line 1087, in create_instrument
    return cls._create(paramset)
  File "C:\Users\eweiss\.conda\envs\IPAEM\lib\site-packages\instrumental\drivers\__init__.py", line 513, in _create
    obj._initialize(**paramset.get('settings', {}))
  File "C:\Users\eweiss\.conda\envs\IPAEM\lib\site-packages\instrumental\drivers\spectrometers\thorlabs_ccs.py", line 159, in _initialize
    self._open(self._address)
  File "C:\Users\eweiss\.conda\envs\IPAEM\lib\site-packages\instrumental\drivers\spectrometers\thorlabs_ccs.py", line 177, in _open
    handle = self._NiceCCSLib.init(self._address, True, False)
  File "C:\Users\eweiss\.conda\envs\IPAEM\lib\site-packages\nicelib\nicelib.py", line 1153, in __call__
    return self.sig.extract_outputs(c_args, retval, ret_handler_args)
  File "C:\Users\eweiss\.conda\envs\IPAEM\lib\site-packages\nicelib\nicelib.py", line 214, in extract_outputs
    retval = self.ret_handler.handle(retval, ret_handler_kwargs)
  File "C:\Users\eweiss\.conda\envs\IPAEM\lib\site-packages\nicelib\nicelib.py", line 537, in handle
    return self.__func__(retval, **kwargs)
  File "C:\Users\eweiss\.conda\envs\IPAEM\lib\site-packages\instrumental\drivers\spectrometers\thorlabs_ccs.py", line 54, in ccs_errcheck
    raise ThorlabsCCSError(NiceCCSLib.error_message(0, error_code)[0])
instrumental.drivers.spectrometers.thorlabs_ccs.ThorlabsCCSError: D
Exception AttributeError: "'CCS' object has no attribute '_NiceCCS'" in <bound method CCS.__del__ of <instrumental.drivers.spectrometers.thorlabs_ccs.CCS object at 0x000000000EEAC7F0>> ignored`

I tried running this with python 3.7 and 2.7

Would appreciate any help that can be provided. 

PS The thorlabs drivers are installed as I can use the device within their software.

natezb, 2019-10-08:
The first issue is that the error message isn't being printed properly. This was fixed in a0296e56e55648770c0f3033b49469d0d797acd0, which hasn't been released yet, so you might try installing the dev version from GitHub. That way you can at least see what error the Thorlabs API is returning.

The AttributeError at the very bottom isn't a big issue (and isn't the source of your problem), but I might make a quick commit to patch that up.

eveloen, 2019-10-11:
Thanks for the quick response.
I pulled the current Git version. When running the same code again, the error message has slightly changed. The new message is as follows.

`Exception instrumental.drivers.spectrometers.thorlabs_ccs.ThorlabsCCSError: ThorlabsCCSError('The given session or object reference is invalid.',) in <bound method CCS.__del__ of <instrumental.drivers.spectrometers.thorlabs_ccs.CCS object at 0x000000000B18BA58>> ignored`

Even with this more meaningful error message, we're still stuck.
Further help would be much appreciated.