mabuchilab/Instrumental #104: PCO Pixelfly USB Initialization fails with ZeroDivisionError

zakv, 2019-12-02:
Hello,

I'm attempting to use instrumental to control a pco.pixelfly USB camera, which is compatible with pco.sdk. However, when I attempt to initialize the camera, I receive a ZeroDivisionError (code and traceback below).

This root issue behind this is that the pco.pixelfly usb doesn't support ROI. The pco.sdk manual mentions that if a camera doesn't support ROI then GetCameraDescription() will return zero for wRoiHorStepsDESC. That means that hstep in PCO_Camera._set_ROI is set to zero, which leads to the ZeroDivisionError on line 309.

From the chart on page 282 of the pco.sdk v1.24 manual, it seems that five of their cameras aren't compatible with the function PCO_SetROI, so squashing this bug should help for other camera models as well.

One workaround may be to simply add "`if hstep == 0: return`" in PCO_Camera._set_ROI and maybe issue a warning. Although maybe it would be better practice to forgo calling PCO_Camera._set_ROI at all if the camera doesn't support it?

Cheers,
Zak

Software Versions:
Windows 10
pco.sdk 1.24
instrumental-lib 0.5
Python 3.7.4
all are 64 bit

Code:
`cd "C:/Program Files (x86)/Digital Camera Toolbox/pco.sdk/bin64/"`
`import instrumental
instrument_list = instrumental.list_instruments(module='cameras')
camera_instrument = instrumental.instrument(instrument_list[0])`

Traceback:
C:\Users\Rubidium\Software\Anaconda3\lib\site-packages\cffi\cparser.py:150: UserWarning: String literal found in cdef() or type source. String literals are ignored here, but you should remove them anyway because some character sequences confuse pre-parsing.
  warnings.warn("String literal found in cdef() or type source. "
#---------------------------------------------------------------------------
ZeroDivisionError                         Traceback (most recent call last)
<ipython-input-2-28dc4a319e8b> in <module>
      2 instrumental.list_instruments(module='cameras')
      3 instrument_list = instrumental.list_instruments(module='cameras')
----> 4 camera_inst = instrumental.instrument(instrument_list[0])

~\Software\Anaconda3\lib\site-packages\instrumental\drivers\__init__.py in instrument(inst, **kwargs)
   1265         inst = find_visa_instrument_by_module(params)
   1266     else:
-> 1267         inst = find_nonvisa_instrument(params)
   1268 
   1269     if inst is None:

~\Software\Anaconda3\lib\site-packages\instrumental\drivers\__init__.py in find_nonvisa_instrument(params)
   1155                              for k,v in params.items()}
   1156         if hasattr(driver_module, '_instrument'):
-> 1157             return driver_module._instrument(normalized_params)
   1158 
   1159         full_params = find_full_params(normalized_params, driver_module)

~\Software\Anaconda3\lib\site-packages\instrumental\drivers\cameras\pco.py in _instrument(paramset)
    680     # Because opening devices is so slow, we override the default implementation which
    681     # calls list_instruments. This instead just tries to open the camera
--> 682     return PCO_Camera._create(paramset)  # FIXME

~\Software\Anaconda3\lib\site-packages\instrumental\drivers\__init__.py in _create(cls, paramset, **other_attrs)
    511         obj._before_init()
    512         obj._fill_out_paramset()
--> 513         obj._initialize(**paramset.get('settings', {}))
    514         obj._after_init()
    515         return obj

~\Software\Anaconda3\lib\site-packages\instrumental\drivers\cameras\pco.py in _initialize(self)
    181 
    182         _, _, max_width, max_height = self._get_sizes()
--> 183         self._set_ROI(0, 0, max_width, max_height)
    184 
    185     # Enums

~\Software\Anaconda3\lib\site-packages\instrumental\drivers\cameras\pco.py in _set_ROI(self, x0, y0, x1, y1)
    307         cx = self.max_width // 2
    308         xdiff = max(cx - x0, x1 - cx) - 1
--> 309         xdiff = (xdiff // hstep + 1) * hstep
    310         fx0 = cx - xdiff
    311         fx1 = cx + xdiff

ZeroDivisionError: integer division or modulo by zero

natezb, 2019-12-14:
Thanks for the thorough report! Unfortunately I no longer have access to this camera, so I can't really test any changes made. If you're willing to test out a fix, I can create a bugfix branch and PR for you to look over.

Having just looked over the code, I see that it takes a 2-layered approach to ROI. First, it sets the hardware ROI to the smallest bounding box allowed, and second it keeps track of what needs to be trimmed off in software to produce the user-requested ROI.

We can use `wRoiHorStepsDESC` as you suggested to see if hardware ROI is supported. If not, we simply use the full sensor as the ROI and don't call `SetROI`, and all of the trimming will be done in software.

zakv, 2019-12-15:
Ah, hmm this might be tough to fix properly without access to your old camera. I actually managed to work around the `_set_ROI()` bug through some monkey patching, but ran into a couple of other issues after that. I'm a little worried that attempting to patch all of those might make instrumental compatible with the pixelfly USB, but accidentally break compatibility with other cameras in the process. I ended up writing a new class to replace `PCO_Camera` which works for my purposes anyway, so I wouldn't mind if this issue gets tagged as "will not fix" and closed. That said, if you'd prefer to try to iron those bugs out, I'd be happy to test and contribute.

On a related note, I'm using instrumental in an attempt to make PCO Cameras work with Labscript, a software package for controlling hardware-timed experiments such as those in AMO laboratories. So far I've kept the code in a private repo for my lab, but I'm hoping to share it and maybe see if the Labscript developers would like to integrate it into their official packages. I've used imports as much as possible to work with Instrumental's code at arms length. However to do some workarounds I had to copy some bits of code to my project to edit them, e.g. to add more NiceLib wrappers for other PCO SDK functions. Would you mind if I shared those edited bits of code? Alternatively I'd also be happy to contribute those changes directly to Instrumental.

natezb, 2019-12-21:
I think it's worthwhile to at least look over the changes that you've needed to make to get your camera working. Ideally this driver would properly support any camera using PCO.SDK. I've created PR #105 (and the `pco` branch) to collect these changes in one place.

As for your second question, it would be good to see the edited bits of code you're talking about. Then we can decide whether it makes more sense to integrate it into Instrumental or do something else instead.

zakv, 2019-12-22:
Great! That's awesome that you're still interested in improving the codebase even though you no longer use this kind of camera. I think putting these changes into Instrumental is the right idea, I just didn't wan't to pressure you into supporting code that's no longer of use to you. I'll write up the further issues I ran into with the camera on the pull request conversation.

natezb, 2020-04-10:
Fixed by #105

closed 2020-04-10 by natezb