mabuchilab/Instrumental #108: redimensioning of widget in gui

Brieussel, 2020-03-25:
Hi there,
first thanks a lot for this extremely useful tool.

I am using the camera in a QtWidget thanks to your script gui.py: using the class `CameraView` which is working as a QLabel.
I would like to redimension it by using my object `QLabelObjet.setFixedSize(500, 500)` or even use your functions` set_height(self, h):` and `set_width(self,w): `
but it is not possible because you are constantly redefining the size at the end of `_set_pixmap_from_array(self, arr):` and it erases my changes.

Wouldn't it make more sens to only make this rescaling one time rather than at each frame?

Brieussel, 2020-03-27:
I solved the problem by commenting the two lines:
 ```
#       if pixmap_size != self.size():
  #          self.setMinimumSize(self.pixmap().size())
 
```
in the function `_set_pixmap_from_array` in gui.py and now i can use `set_height` and `set_width` without any problems.

natezb, 2020-04-01:
Thanks for reporting this, and investigating the cause. I just pushed f5d1a45, which adds an `autoresize` parameter which lets you disable this resizing.

closed 2020-04-01 by natezb

Brieussel, 2020-04-01:
exactly what i needed, thanks for the push :)

Alexandre