mabuchilab/Instrumental #109: making a .exe with instrumental

Brieussel, 2020-03-27:
Hi everyone,

I tried desperately to create a .exe file with python programs that import instrumental but it always failed. I tried with cx_freeze and pyinstaller, but no success until now.
cx_freeze always give me a `fatal python error initfsencoding unable to load the file system code`
and with pyinstaller it just opens and crash.
Does anyone already succeed to do that and would be kind enough to share his or her setup.py?

I would help me a lot

Brieussel, 2020-04-01:
I find a way to do it so I answer to my own question in case someone get the same problem.
I made it work with pyinstaller. First I would recommend not to use anaconda, but a classical python installation or pyinstaller will try to install a lot of modules and crash. Then make sure you pip install all the important modules  numpy, nicelib, pywin32 ,pint,pyqt5,scipy,qtpy (hope I didn't forget one).
And finally you need to add the implicit libraries import that nicelib does but that pyinstaller can't find. In my case for DC thorlabs camera i needed to import instrument in this way: from 
```

from instrumental import instrument, list_instruments, gui	#initialise instrumental the normal way
import instrumental.drivers.cameras.uc480	# make sure the includes are taken by pyinstrumental
import instrumental.drivers.cameras._uc480lib	#make sure that the dll link is made in pyinstrumental

```

natezb, 2020-04-01:
Thanks for providing this information!

closed 2020-04-01 by natezb