mabuchilab/Instrumental #112: Buffer issue with python 3.8 and thorlabs CCD spectrometer

grahamjtr, 2020-04-09:
Hi,
I've seen two other issues reporting this buffer issue but following them does not seem to fix my issue.

I have installed instrumental to python 3.8 using git to clone the github repository.
I am able to load_instruments, (though I get a warning about Sig):

>>> from instrumental import instrument, list_instruments, u
>>> paramsets = list_instruments()
C:\Users\Asus\AppData\Local\Programs\Python\Python38-32\lib\site-packages\nicelib\nicelib.py:564: FutureWarning: Using tuples to define signatures has been deprecated in favor of using Sig. See the documentation for more information.
  sigs.update((n, (v if isinstance(v, Sig) else Sig.from_tuple(v)))
>>> paramsets
[<ParamSet[CCS] usb='USB0::0x1313::0x8089::M00569074::RAW' serial='M00569074' model=<SpecTypes.CCS200: 32905>>]

But then I get the buffer error when I execute instrument:
>>> instrument(paramsets[0])
>>> ...(rest of the error is removed, let me know if you need it)
File "C:\Users\Asus\AppData\Local\Programs\Python\Python38-32\lib\site-packages\instrumental\drivers\spectrometers\thorlabs_ccs.py", line 293, in _cdata_to_numpy
    buf = buffer(ffi.buffer(ffi.addressof(cdata), size)[:])
NameError: name 'buffer' is not defined

I have it working ok in python 2.7, but would like it working in 3.8 if possible
Thanks for any help

natezb, 2020-04-09:
It looks like that code isn't using the latest git version, did you remember to install it after cloning the repository? Or perhaps you installed the git version under Python 2 but not Python 3.

If you're still having problems, maybe an old install is interfering, so you might `pip uninstall instrumental` first.

grahamjtr, 2020-04-15:
Thanks so much, you were right, I had an old install interfering with it :-)

closed 2020-04-15 by grahamjtr