mabuchilab/Instrumental #114: Exception when closing Thorlabs CCS spectrometer

graham-triggs, 2020-05-13:
I'm getting this exception when my script finishes. I am successfully closing the spectrometer with the close() method (I think), as this error only appears after my whole script finishes. As it happens on closing it isn't a huge problem but I'd like to get it working correctly.
I'm using pyton 3.8 32bit and cloned the latest instrumental version. Everything else with using the spectrometer via instrumental seems to be working fine.
Thanks for any help!

Exception ignored in: <function CCS.__del__ at 0x12C646A0>
Traceback (most recent call last):
  File "C:\Users\Asus\AppData\Local\Programs\Python\Python38-32\lib\site-packages\instrumental_lib-0.6.dev0-py3.8-win32.egg\instrumental\drivers\spectrometers\thorlabs_ccs.py", line 170, in __del__
  File "C:\Users\Asus\AppData\Local\Programs\Python\Python38-32\lib\site-packages\instrumental_lib-0.6.dev0-py3.8-win32.egg\instrumental\drivers\spectrometers\thorlabs_ccs.py", line 189, in close
  File "C:\Users\Asus\AppData\Local\Programs\Python\Python38-32\lib\site-packages\nicelib\nicelib.py", line 1133, in __get__
  File "C:\Users\Asus\AppData\Local\Programs\Python\Python38-32\lib\site-packages\nicelib\nicelib.py", line 684, in __init__
  File "C:\Users\Asus\AppData\Local\Programs\Python\Python38-32\lib\site-packages\nicelib\nicelib.py", line 687, in _assign_signature
ImportError: sys.meta_path is None, Python is likely shutting down

natezb, 2020-05-18:
Thanks for the report. Can you test if the error still occurs if you remove your call to `close()`? I'm wondering if this is because of a double-close or a more specific issue with `__del__()`.

aaronscode, 2021-04-01:
Just to add testament from a second person, I'm getting the same issue. I can recreate it with the following code:

```
from instrumental import instrument, list_instruments

paramsets = list_instruments(module='spectrometers')
daq = instrument(paramsets[0])

intensity, wavelength = (daq.take_data())
print(intensity)
```

which yields the following exception (basically the same as @graham-triggs):

```
Exception ignored in: <function CCS.__del__ at 0x00000222B1E0DE50>
Traceback (most recent call last):
  File "C:\Users\aaron\AppData\Local\Programs\Python\Python39\lib\site-packages\instrumental\drivers\spectrometers\thorlabs_ccs.py", line 171, in __del__
  File "C:\Users\aaron\AppData\Local\Programs\Python\Python39\lib\site-packages\instrumental\drivers\spectrometers\thorlabs_ccs.py", line 190, in close  
  File "C:\Users\aaron\AppData\Local\Programs\Python\Python39\lib\site-packages\nicelib\nicelib.py", line 1140, in __get__
  File "C:\Users\aaron\AppData\Local\Programs\Python\Python39\lib\site-packages\nicelib\nicelib.py", line 689, in __init__
  File "C:\Users\aaron\AppData\Local\Programs\Python\Python39\lib\site-packages\nicelib\nicelib.py", line 692, in _assign_signature
ImportError: sys.meta_path is None, Python is likely shutting down
```