mabuchilab/Instrumental #117: Using simulated Thorlabs Kinesis instruments with Instrumental

burggraaff, 2020-06-29:
Hi,

I'm working on some software to interface with a Thorlabs TDC001 cube, among other instruments. My lab access is currently very limited because of COVID-19, so for now I'm working only with simulated instruments, using Kinesis.

I've installed Instrumental and Kinesis, and it has succesfully created the `drivers.motion.kinesis` and `drivers.motion.tdc_001` modules. However, when I run `list_instruments()`, with a simulated TDC001 cube running in the background, the return is simply an empty list.

Furthermore, when I try to manually create a TDC001 object (`tdc1 = instrumental.drivers.motion.tdc_001.TDC001()`), I get the following error: `Exception: <ParamSet[TDC001] > does not match any known ParamSet from driver module motion.tdc_001.` I'm not sure what to make of this.

Kinesis itself can see the simulated TDC001, so I think the problem is in the Instrumental interface. Does Instrumental not (currently) support simulated instruments at all, or is something going wrong?

Many thanks,
Olivier

natezb, 2020-06-30:
Thanks for the report.

I just tried using our NiceLib-based interface to Kinesis to interact with a simulated device and can confirm that it does not work. `GetDeviceListSize` returns zero, `GetDeviceListExt` returns an empty string, and `Open` fails to see the device.

Before spending much more time poking around, it may make the most sense for you to contact Thorlabs support and ask if the Kinesis *C API* supports talking to simulated hardware. It may be that they only support it through the Dot NET API, which Instrumental does not use.

If talking to simulated devices *is* supported by the C API, it's likely that we can fix this issue.

burggraaff, 2020-07-01:
Thank you for the quick reply. I have contacted Thorlabs support, but have not received a response yet.

In the mean time, I tested my installation with a pair of real TDC001 cubes. Instrumental can detect them if I am _not_ running Kinesis, but cannot see them if Kinesis is running at the same time. Is this known behaviour? It might be due to similar reasons as the problem discussed in the original post.

burggraaff, 2020-07-03:
I got a reply from Thorlabs support asking whether Instrumental uses the API command described in the attachment.

[image]

natezb, 2020-07-10:
Sorry, somehow I missed your reply. No, we definitely don't use that in the current code. That's good news, seems we should be able to get simulations working in Instrumental! I'll try to find some time this weekend to look into implementing support.

natezb, 2020-07-15:
Ok, I am able to see a simulated device now when using `InitializeSimulations()`. This might be a bit tricky to integrate into our general kinesis driver (since `Thorlabs.MotionControl.DeviceManager.dll` appears not to contain this function), but for now it shouldn't be too hard to integrate into the older `tdc_001` driver that you're using.

There's also the question of how/when simulations should be enabled. To start with we can add init/uninit functions to the top level of the `tdc_001` module. You'd then need to import that module directly and enable simulations before `list_instruments()` would show the simulated instruments.

In the future, maybe we could add a config option to tell the driver whether or not it should look for simulated instruments.

natezb, 2020-08-25:
I just pushed some commits (c1dc2d5d909614d272c039fd2a658db2412b173e, 3ec8eed4ee7558afca39d9c16587c8ba68bfc72e) that address this issue, for both the `TDC_001` and generic `kinesis` drivers. Each module has a convenience function to connect to the kinesis simulator:

```
from instrumental.drivers.motion import tdc_001
tdc_001.initialize_simulations()
tdc_001.list_instruments()
```

If you actually want to create a `TDC001` device, things are a bit trickier. It appears that `CC_LoadSettings` fails in the init function, leaving the "real world unit" conversion factors unset. You could probably set these factors manually if you want to use the position-related functions.

closed 2020-08-25 by natezb