mabuchilab/Instrumental #118: Remote server set-up not functional with Python 3

mankapandey, 2020-08-20:
I'm able to start a remote server using Python 2.7 (miniconda2) and the provided instr_server.py file. However, when I attempt to start a remote server with the same file in Python 3.8 (miniconda3), I get the following error message:

`  File "instr_server.py", line 10, in <module>`
`      from instrumental.drivers.remote import ThreadedTCPServer, DEFAULT_PORT`
`  File "C:\..\miniconda3\lib\site-packages\instrumental\drivers\remote.py", line 13, in <module>`
`      import cPickle as pickle`
` ModuleNotFoundError: No module named 'cPickle'`

Would it be possible to provide an instr_server.py file that works with Python 3, or can you recommend any workarounds? Thanks in advance.

natezb, 2020-08-20:
Thanks for the report. I just tested it, and the latest git version of Instrumental's server system appears to support Python 3, so I'd suggest upgrading and giving it a try.

One more reason I should get around to packaging a new release...

mankapandey, 2020-08-20:
Thanks for checking! Before I saw your comment, I was able to get it working by running 2to3 on remote.py.

natezb, 2020-08-20:
Ok cool, I'll close this then. If you have any more issues, try out the git version as there are a number of things that have been fixed but not yet released

closed 2020-08-20 by natezb