mabuchilab/Instrumental #134: Thor Labs CCS200 - Constantly Timing Out (Ive seen the other closed thread on this)

tz15, 2021-02-11:
Hi there, using an anaconda dist with everything installed and constantly receiving the ThorLabsCCSError: b'Timeout expired before operation completed" even when I choose incredibly fast integration times the timeout still occurs.

natezb, 2021-02-12:
Hi, thanks for the report. I'm not sure specifically which issue you're referring to, but have you seen #120? By the end of the thread, @LewisBarltrop came up with a theory of what the problem might be, related to a low-power USB state.

tz15, 2021-02-12:
Thank you for the reply! I have seen #120 as well as #93, have tried many USB cables and both desktop/laptop. Seems to be timing out no matter what, working maybe 1/100 times? Thank you for making all of this, super helpful!!

tz15, 2021-02-22:
Just investigated the USB power, and the spectrometer stays at USB D0, full power, the entire time.

tz15, 2021-03-12:
Could not fix this issue, was able to successfully collect data reliably with pythonNET. [ThorLabsCCS200](https://github.com/tz15/ThorLabsCCS200)

closed 2021-03-12 by tz15

natezb, 2021-03-12:
Sorry I couldn't be of more help, I don't have much experience with this instrument. Thanks for posting about pythonNET, we could perhaps create a .NET-based driver if other people continue to have issues and the .NET lib seems to be more reliable. We'd probably want to keep this driver too though, for cross-platform support.

tz15, 2021-03-12:
No worries, it was a tricky workaround with lots of time spent talking to ThorLabs. Thank you for making this entire library, it is amazing. Happy to help however I can!