mabuchilab/Instrumental #135: A way to open CS165MU thorlabs camera

alexsludds, 2021-03-27:
Hello,
I have a CS165MU camera from thorlabs and I want to grab frames from the camera in real time and process them on the fly.
I noticed that there is support for the TSI SDK in instrumental, but it's not obvious to me how to use it.
Do you have any advice on how I can use this camera?

natezb, 2021-03-30:
You should be able to use it like any of the other camera drivers. To aid in debugging, you might want to import that module directly and look for any attached cameras:
```
from instrumental.drivers.cameras import tsi
tsi.list_instruments()
```
The camera [can then be opened](https://instrumental-lib.readthedocs.io/en/stable/instruments.html) and used like the other cameras. It looks like the latest docs for the camera interface haven't generated properly, so check out [these old ones](https://instrumental-lib.readthedocs.io/en/0.5/cameras.html).

Please note that the `tsi` driver is a bit experimental because it wraps a C++ library in a hacky way. However, it worked well enough for me several years ago.

alexsludds, 2021-03-30:
Thank you so much! This has been helpful to know this will work.
Can you point me to a location where I can download the TSI SDK? I downloaded the current version of thorcam, but it seems to only come with the compact scientific camera sdk. Also, do I have to do anything special once I have to C++ SDK files to get the python to see them?

Thank you again,
-Alex

natezb, 2021-04-01:
It's been years since I used this, but it appears that the TSI software is available if you go to the software page where you'd download ThorCam, and go to the "Programming Interfaces". It should be under "Windows SDK and Doc. for Scientific Cameras".

Interestingly, it seems like they've expanded the SDK over the years and now have a C interface which we could wrap more cleanly. Also they seem to have a Python wrapper of their own in that same download, if you're interested.

natezb, 2021-04-01:
As far as installation goes, I think the Instrumental driver only relies on `tsi_sdk.dll` being accessible via your PATH.

In all honesty though, unless you *really* want to use Instrumental for this specific driver, you're probably better off using the Thorlabs-provided Python interface, since it seems like the SDK may have changed significantly since I was using it. In fact, `tsi_sdk.dll` is within the "Legacy" subdirectory of the SDK zip file.