mabuchilab/Instrumental #137: Driver "out of date" message

bws5, 2021-04-01:
Hello, I am trying to use a Thorlabs DCx camera with instrumental (using python 2.7 in Windows 7). Instrumental sees the camera, but I get the following message:

> UserWarning: Driver 'instrumental.drivers.motion.ecc100' is out of date and incompatible with the current Instrumental core. Pull requests fixing this are welcome.

Could this be related to the sunset of python 2, or is there a way around this issue on my current system?

Thanks!

Ben

natezb, 2021-04-01:
No, this isn't related to the end of Python 2 support (though it is worth noting that future releases of Instrumental will no longer be sticking to support for Python 2). This notice has nothing to do with the DCx driver and shouldn't affect you at all. I added this message to signal that the `motion.ecc100` driver is out of date to anyone who might use it. However, since this gets called for *everyone* upon a top-level `list_instruments()`, that was probably a bad decision and should be reversed somehow.

If you really care to, you may be able to suppress the warning through some Python settings.

bws5, 2021-04-01:
Thanks for the quick reply, Nate!

closed 2021-04-02 by bws5

natezb, 2021-04-02:
You're welcome!